HclX/WyzeHacks #84: Wyzecam v2 on latest retail FW won't pull firmware.bin with remote_install.sh

denneyliptak, 2021-02-10:
Running the latest release of wyzehacks (0.5.01) and am stuck at as the camera doesn't appear to request the firmware update. 
My v2 camera is running the most recent retail FW (4.9.6.199)

If I issue a request from another machine on the same network (ie. wget http://10.10.X.X:11808/firmware.bin) I see the  _"GET /firmware.bin HTTP/1.1" 200_ status update from the script, which suggests that the wyze_updater script has successfully started the webserver and that it is reachable from inside my network.

Can you recommend any suggestions for troubleshooting? The camera has an SD card inserted; if there's a manual install involving the card, that'd work too. But I'd like to get the remote updates enabled for the future.

HclX, 2021-02-12:
I've verified this firmware and the installer works as expected. That being said, I've seen my camera on a different version sometimes not fetching update issued by wyzeupdater. The way it works is we push a request to the wyze server, and then wyze server pushes the request down to the camera. So it's possible something wrong on the server side, but since this is a step before our script is running, there is not much  we can do debugging the process.

I'd suggest you try a firmware recovery using SD card to a different firmware and see if that works.

denneyliptak, 2021-02-13:
Thank you for the reply! I apologize if the question is particularly noob but is there a repo somewhere with Wyze FW? I signed up for their beta program hoping that might expose something but I'm not seeing download locations for some of the newer-than-199 FW that folks have been reporting on.

HclX, 2021-02-16:
@denneyliptak It looks like they have pulled later firmwares due to problems, 199 firmware should still work.

closed 2021-02-21 by HclX