HclX/WyzeHacks #85: Wyze Cam Pan telnet password does not work

jakepals, 2021-02-16:
I tried using the same password I use for the v2 cameras, but it looks like the password is incorrect. Is there a different password for the pan cameras? 
```
~/wyze_hacks_0_5_01 $ telnet 192.168.x.x
Trying 192.168.x.x...
Connected to 192.168.x.x.
Escape character is '^]'.
PanCam login: root
Password:
Login incorrect
PanCam login:
Login timed out after 60 seconds
Connection closed by foreign host.
```

revenant-81, 2021-02-16:
AFAIK, the Cam Pan password is unknown. Within config.inc, uncomment the PASSWD_SHADOW variable and push the config to the Pan camera. This will update the password to the same as the V2s.

jakepals, 2021-02-16:
Uncommenting the line works. Thank you. I am able to telnet in.

closed 2021-02-16 by jakepals