HclX/WyzeHacks #87: Must insert real sd card after firmware update

hyukishi, 2021-02-20:
While this wasn't a huge issue for me, I was able to install Wyze hacks on each v3 camera I own. Afterwards, I updated each one to the latest official firmware. WyzeHacks persisted as expected, but I had to insert a physical sd card into each camera long enough for it to register before the cameras could resume recording to the NFS share on my server.

HclX, 2021-02-21:
I think i saw this symptom on the new firmware: they have changed how to detect sd card again, will work on a fix for this.

HclX, 2021-02-21:
I just pushed a fix to the latest `bugfix` branch and verified it with v3 firmware version 252. Please test if it solves the problem.

hyukishi, 2021-02-21:
Verified working on all 4 of my v3 cameras. Thanks for the quick turnaround! Prior to applying the bugfix, rebooting the cameras was also resulting in the NFS shares being unmounted without a physical sd card present in each camera. Now that is a thing of the past!

closed 2021-02-22 by HclX

PDXUser, 2021-03-05:
Would this also apply to Wyze Firmare 4.10.6.199?  I noticed that the camera rebooted a few times last week, and the hack was gone.  I will put a SDCard in this weekend if needed, but I find it odd that it nuked the install automatically.

HclX, 2021-03-05:
V2 is different than v3. On v2 sometimes the hack gets wiped out by update
so you will have to install it again.