HclX/WyzeHacks #91: Videos not being saved on NFS share

rfg81, 2021-02-28:
Hi, I'm probably doing something wrong but please bare with me. I can see files being saved on the NFS share folder. I'm using a windows 10 NFS share by using this command:

winnfsd.exe -id 0 0 -log on -addr 192.168.10.10 D:\Camera\Wyze /camera

The camera is Wyze V2 running 4.9.6.199 and depending on my luck sometimes I can see SD card memory available as 0MB/ 0MB and sometimes it doesn't detect an sd card. I tried multiple reboots and multiple remote_install (version 0.5.03).

I can see a config.inc file on the windows machine with this information:

# USER CONFIGURATIONS -- TO BE MODIFIED BY USER
# Make your own copy of this file and put it under the root of SD card
# with name "config.inc" when installing the hack.

# You need to modify NFS_ROOT to point to your NFS share. The recordings
# will be stored under $NFS_ROOT/WyzeCams/<MAC_ADDR_OF_YOUR_CAMERA>.
export NFS_ROOT='192.168.10.10:/camera'

# Some NFS configurations may need special mount options. You can specify
# them here. The default value is verified on my camera with a NFS v3 share
# on my home server.
# export NFS_OPTIONS='-o nolock,rw,noatime,nodiratime'

# Timeout threshold for NFS connectivity check. If the NFS share is not
# available for more than '$NFS_TIMEOUT' seconds, a reboot will be initiated
# trying to resolve the network connectivity. The default value is 15 seconds.
# export NFS_TIMEOUT=15

# Uncomment and modify the AUTO_REBOOT variable to automatically reboot
# the camera at a specific minute. Time is specified in "HH:MM" format
# in local time.
export REBOOT_AT='02:15'

# Uncomment and modify the ARCHIVE_OLDER_THAN variable to automatically
# archive recording clips and alarm images if it's a certain days older.
export ARCHIVE_OLDER_THAN=5

# Uncomment the following variable to sync the console log to the camera
# folder in NFS share. Use with caution as the log contains sensitive
# information and will cause your account compromise if leaked.
# export SYNC_BOOT_LOG=1

# In my setup I noticed at some occasions the camera starts rebooting frequently
# which seems to be a result of dropping wifi connections. Running a "ping"
# command in background seems to be solving this issue. Uncomment this variable
# to enable this fix.
export PING_KEEPALIVE=1

# Uncomment to enable voice notification when NFS network connection lost. The
# value should be in range of 0 and 100
# export NOTIFICATION_VOLUME=80

# Uncomment and update this value to customize the hostname of the camera. By
# default it will be set to <WyzeCam>-<last 4 hex digits of the mac>.
export HOSTNAME='<Garage>'

# Uncomment this to enable automatically updating config.inc file based on file
# <camera_folder>/wyzehacks/config.new. This will be checked every one minute.
export AUTO_CONFIG=1

# Uncomment this to enable automatically updating wyzehacks. To update, you need
# to extract the latest wyzehacks release archive to $UPDATE_DIR and name it as
# "release_?_?_??". The updating script will search for latest version based on
# the folder name every one minute.
export AUTO_UPDATE=1

# By default, variable "UPDATE_DIR" is set to "/mnt/WyzeCams/wyzehacks", which
# is <NFS_ROOT>/WyzeCams/wyzehacks. If auto update is enabled, it will search
# for updates in this directory. The default value allows all cameras share the
# same update. You can customize this directory for a specific camera if you
# want it to use different update (for example, testing dev builds). Some thing
# under "/media/mmc" would be a good candidate because it maps to the camera
# folder.
# export UPDATE_DIR='/media/mmc/wyzehacks'

# Uncomment and update this to run a custom script after the NFS mount finishes.
# The script must already exists and executable. You can always use "/media/mmc"
# to refer the camera folder on the NFS share. Or use "/mnt" to refer the root
# of NFS share. The value of the variable can only contain script file name, no
# command line arguments. The script will be executed in background.
# export CUSTOM_SCRIPT='/media/mmc/scripts/myscript.sh'
# export CUSTOM_SCRIPT='/mnt/WyzeCams/scripts/myscript.sh'

# !!!ADVANCED USER!!!
# This changes the default root password to whatever you specified. To
# generate your own password hash, use the following command from a
# linux environment:
#   openssl passwd -1 -salt <YOUR SALT> <YOUR PASSWORD>
# Use the result to replace sample hash ($1$MYSALT$3Sy1OLRk4kTa7P6fvzwp71)
# export PASSWD_SHADOW='root:$1$MYSALT$3Sy1OLRk4kTa7P6fvzwp71:10933:0:99999:7:::'

In the log folder I see several reboot_XX files but all have 0 kb.

Could you please tell me what I did wrong?

HclX, 2021-03-13:
i don't see any problem in your config file, and it seems the camera is correctly mounting the NFS share and storing logs already. Those reboot_XX files are expected to show you when the camera has rebooted. The latest wyze hack is very sensitive to the running firmware version so beta firmwares are not well supported. I'd suggest you reflash your camera firwmare using SD card recovery method and then try it again.

HclX, 2021-03-27:
@rfg81 can you sd recover the firmware to latest version (241) and try hack version 0.5.04?

ZuluSpl0it, 2021-05-11:
I'm seeing the same thing as rfg81 using a v2 camera on firmware 241 with 0.5.04. Constant reboots when it connects to the NFS share. When the camera doesn't find the NFS share it actually doesn't reset so quickly so I am able to telnet into the camera.

ZuluSpl0it, 2021-05-14:
Ok, switched to running a different NFS and that seems to have corrected the constant reboots.

I followed this guide to setup a NFS: https://pimylifeup.com/raspberry-pi-nfs/

However, videos were still not being saved. Only jpgs in the "alarm" folder were being saved. The permissions for the "record" folder were different than the 'alarm' folder. I did a
`cd /media/mmc`
`chmod 755 record`
from the camera via telnet and now it saves the mp4s. I have a v2 running WyzeHacks 0.5.06

ZuluSpl0it, 2021-05-14:
I confirmed that it is a permissions issue. The permissions are set as drw-r--r-- when the camera makes a new folder for the next hour.
[image]

My if I change the permissions via the raspberry pi GUI:
[image]
 or issue the following commands via telnet to the camera:
`cd /media/mmc
chmod 777 record`
[image]

the mp4s are saved as expected.

I thought of running a script on the pi to change the permissions of the newly created folders, but I think it would be a better fix to have the camera create the folder with the drwxrwxrwx permissions.

HclX, 2021-05-14:
Might be something issue with the umask. Can you try if the follow command
fixes the issue?

Telnet into your camera
Run "echo 'umask 000' >> /configs/wyze_hack.cfg"

ZuluSpl0it, 2021-05-14:
The permissions are now:
[image]

Still not writing mp4s to the NFS until I chmod 777 on the folder

HclX, 2021-05-16:
@ZuluSpl0it wait, your directory owner/group is not root but 1000?

ZuluSpl0it, 2021-05-16:
My NFS options used are:
`/mnt/nfsshare *(rw,all_squash,insecure,async,no_subtree_check,anonuid=1000,anongid=1000)`

So the anonymous login from the camera should use pi's UID and GID (1000).

holocronology, 2021-05-17:
> How are others running the NFS server?

I'm not. Like you, it never worked, either via windows NFS or a physical PI that I set up. Gave up...

Given the improved resources on the v3, I'd advocate for adding SMB to the v3 install.

ZuluSpl0it, 2021-05-24:
I ran the NFS server on a real raspberry pi 3 (unlike the virtualized Raspberry Pi Desktop for PC above) and the camera records without any permissions issues. The folders and mp4 get created as expected. Does anyone know what causes the difference between the Raspberry Pi Desktop for PC and the armhf version for the real hardware?

delovelady, 2021-06-07:
I had this issue on a freenas system when I had set export permissions to mapall group and mapall user.  Once I removed those settings at the server, the issue went away.

[image]

twig123, 2021-07-21:
> > How are others running the NFS server?
> 
> I'm not. Like you, it never worked, either via windows NFS or a physical PI that I set up. Gave up...

I have a post, that I made with info on getting the Hanewin NFS Server working on Windows:
#51