HclX/WyzeHacks #93: I installed WyzeHacks on my Wyze Doorbell...

hyukishi, 2021-03-01:
And it installed successfully using only the remote_install.sh. However, since the doorbell doesn't have an option to use an SD card, there are no recordings from it going to my NFS share. There's also no option to enable SD recording for the doorbell in the app (which was to be expected).

Updating the firmware to the latest official did not yield any sort of persistence with the hack.  I was able to reinstall the hack to the doorbell post firmware upgrade.

I know the doorbell isn't officially supported, but I wanted to pass along the pertinent information in hopes that it will help with future development.

HclX, 2021-03-03:
yes, the wyzehack can be installed on doorbell, but there is not much you get other than telnet access: the doorbell doesn't store any recordings so no way to redirect it to nfs. Still, there might be some possibilities but i'm really busy recently  and couldn't spent any time on the hack.

closed 2021-04-25 by HclX

ErrorErrorError, 2022-08-25:
would it be possible to add the ability to get the rtsp of the stream and or send events to other devices with telnet?