HclX/WyzeHacks #94: WyzeCam v3: cp: can't create '/system/init/app_init_orig.sh': Read-only file system

denneyliptak, 2021-03-02:
Hello! This attempt is being made on a recent v3 camera shipment (ordered Feb 22). I've upgraded camera firmware to 4.36.0.252. I'm using WyzeHacks v0.5.03 and the remote_install.sh script.

The symptom is that I don't see anything writing to the NFS share and telnet doesn't appear to be enabled. (I've been successful with a v2 camera, as a data point). The camera does appear to powercycle when it receives the WyzeHacks code.

Can you please see the install.log (gathered from the SD card post-attempt) pasted below, particularly the section on copying wyze_hack.sh, creating the app_init.sh backup and symlink creation?

Thanks!

**install.log**
```
+ echo WyzeHack: Starting wyze hack installer...
WyzeHack: Starting wyze hack installer...
+ play_sound /tmp/Upgrade/install_snd/begin.wav 50
/tmp/wyze_hack/install/main.sh: line 1: can't create /sys/class/gpio/gpio63/value: nonexistent directory
+ echo 1
+ /tmp/wyze_hack/install/bin/audioplay /tmp/Upgrade/install_snd/begin.wav 50
/tmp/wyze_hack/install/main.sh: line 1: can't create /sys/class/gpio/gpio63/value: nonexistent directory
+ echo 0
+ [ -f /media/mmc/debug/.copyfiles ]
+ echo WyzeHack: Enabling telnetd...
WyzeHack: Enabling telnetd...
+ busybox telnetd
+ set_passwd root::10933:0:99999:7:::
+ /bin/umount /etc
umount: can't umount /etc: Invalid argument
+ rm -rf /tmp/etc
+ cp -r /etc /tmp/
+ echo root::10933:0:99999:7:::
+ /bin/mount -o bind /tmp/etc /etc
+ [ -z 4.36.0.252 ]
+ echo WyzeHack: Current Wyze software version is 4.36.0.252
WyzeHack: Current Wyze software version is 4.36.0.252
+ echo WyzeHack: Installing WyzeHacks version 0.5.03
WyzeHack: Installing WyzeHacks version 0.5.03
+ [ -f /params/wyze_hack.cfg ]
+ [ -f /tmp/Upgrade/config.inc ]
+ [ -f /media/mmc/config.inc ]
+ echo WyzeHack: Use config file /media/mmc/config.inc
WyzeHack: Use config file /media/mmc/config.inc
+ sed s/\r$// /media/mmc/config.inc
+ [ ! -f /configs/wyze_hack.cfg ]
+ [ -z V3 ]
+ echo WyzeHack: Copying wyze hack binary...
WyzeHack: Copying wyze hack binary...
+ cp /tmp/Upgrade/wyze_hack.sh /configs/wyze_hack.sh
+ echo WyzeHack: Hooking up boot script...
WyzeHack: Hooking up boot script...
+ hook_init
+ grep wyze_hack.sh /etc/init.d/rcS
+ [ ! -f /configs/wyze_hack.sh ]
+ [ ! -f /configs/wyze_hack.cfg ]
+ local SYSTEM_DIR=/system
+ [ ! -L /system/init/app_init.sh ]
+ cp /system/init/app_init.sh /system/init/app_init_orig.sh
cp: can't create '/system/init/app_init_orig.sh': Read-only file system
+ readlink /system/init/app_init.sh
+ local APP_INIT=
+ [  != /configs/wyze_hack.sh ]
+ ln -s -f /configs/wyze_hack.sh /system/init/app_init.sh
ln: /system/init/app_init.sh: File exists
+ return 0
+ play_sound /tmp/Upgrade/install_snd/finished.wav 50
/tmp/wyze_hack/install/main.sh: line 1: can't create /sys/class/gpio/gpio63/value: nonexistent directory
+ echo 1
+ /tmp/wyze_hack/install/bin/audioplay /tmp/Upgrade/install_snd/finished.wav 50
/tmp/wyze_hack/install/main.sh: line 1: can't create /sys/class/gpio/gpio63/value: nonexistent directory
+ echo 0
+ rm /media/mmc/version.ini.old
+ mv /media/mmc/version.ini /media/mmc/version.ini.old
+ /sbin/reboot
```

gtxaspec, 2021-03-02:
did you run the v3_init script? also I believe you have to downgrade the firmware to run the unit script.

denneyliptak, 2021-03-02:
Ahh, sorry. Let me go through the readme.md under the v3 directory, I missed that first time around. I'll update/close the github 'issue' shortly.

denneyliptak, 2021-03-02:
All good! Thank you for pointing me in the right direction, greatly appreciated!
Here are my notes, should someone in a similar situation benefit:

1. Get the 4.36.0.228 Factory firmware from Wyze. Unzip, **rename to**: demo_wcv3.bin and place on card.
2. Unplug camera, insert card, hold ‘Setup’ button and reconnect power. Hold ‘Setup’ until light is RedBlue and release. Wait a few min for update and reboot. Check FW level in app and ensure it shows as 228. (this won’t clear the camera out of the wyze app, which is nice)
3. Get the [special init firmware](https://github.com/HclX/WyzeHacks/tree/master/tools/v3/firmwares/4.36.0.228), clear off card and place on card
4. run v3_init.sh (it should report 228 FW on the camera!)
5. After camera reboots, take card out and inspect v3_init.log: Should see a bunch of lines about erasing, writing, verifying then ‘done, rebooting...’
6. Clear off card and place the (customized) config.inc back on the card. Insert card in camv3, wait for the chime (few seconds)
7. Run remote_install.sh (don’t quit script until the camera is done and rebooting)
Attempt telnet to IP or observe files being stored on NFS share.

closed 2021-03-02 by denneyliptak