HclX/WyzeHacks #95: v3: Disable icamera & stream video via open standard?

lachesis, 2021-03-03:
Is there any way to disable the iCamera process and stream video directly to another server without internet access? I've seen a hack based on inotify on the nfs share, but it seems to introduce 60 seconds of latency. I'd need near-realtime streaming (<5 seconds of latency) for my application. The camera hardware is great, but the software, app, and required internet access leaves a lot to be desired.

endertable, 2021-03-05:
@lachesis What you are looking for is probably beyond the scope of this hack, but you can search for 'wyze openmiko' on Google which is an off cloud firmware that has streams for RTSP (<3 seconds) and MJPEG (<1 second) and provides saving video to SD card, ssh access, and other services.  If your server can get the video directly from those http streams then you are good to go with real time.  If you are preferring to use the mp4 files that Wyze creates, then you will be stuck with iCamera and the 60 second delay deal.

HclX, 2021-03-13:
yep, this hack is created to ADD extra functionality while KEEPING the firmware's builtin features. Disabling icamera and enabling RTSP is really out of scope. there are other open source project such as Dafanghacks capable of doing that i believe.

closed 2021-03-13 by HclX

lachesis, 2021-03-16:
Alas, right now neither Dafanghacks nor Openmiko supports Wyze v3. I did some brief investigation and found that v3 does not use a V4L driver, but rather a custom kernel module that talks directly to the camera chip. For the moment, I've put my Wyzecams in a box and bought some other hardware (Amcrest brand) that supports RTSP out of the box.