HclX/WyzeHacks #97: Firmware 4.10.6.199 does not enable SD Card emulation and more

PDXUser, 2021-03-06:
Device type:      Camera (WYZECP1_JEF)
Device name:      Living Room Cam
Firmware version: 4.10.6.199
IP Address:       192.168.1.227

After getting a series of consistent reboots every few minutes and the loss of NFS uploads because the Virtual SD Card went away in the Wyze app, I factory reset the Camera (Wyze Cam Pan) and after reconnecting to the app, I ran the remote install script.  Instead of hearing it talk to me about the flash process like it usually does, the speaker was not making any sound.  

I have the cam next to me right now and I can hear it clicking like the camera is attempting to focus a few times, then it reboots.   

I have not yet figured out why it is rebooting on schedule, I am using the same config as the lower version hacks and it never rebooted unless I manually did so.    In looking through Dmesg, I found a curious entry:

CPU0 RESET ERROR PC:801ECCA0

This was right after the initial entry so I think it may be a false error by chance.  

[root@WyzeCam-0DF6:~]# dmesg
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.10.14 (shizhong@shizhong-pc) (gcc version 4.7.2 (Ingenic r2.3.3 2016.12) ) #4 PREEMPT Tue May 26 13:03:48 CST 2020
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 RESET ERROR PC:801ECCA0
[    0.000000] [<801ecca0>] __delay+0x0/0x10
[    0.000000] CPU0 revision is: 00d00101 (Ingenic Xburst)
[    0.000000] FPU revision is: 00b70000

I can not manage to stay logged in long enough to collect any logs, but there is no "last" command to determine the reboot times so I can get a more accurate time period behind the reboots and not sure if there is an event log which I can view to see what is causing it.

HclX, 2021-03-06:
I don't think that error is relevant. I'd try reflash (not factory reset) your camera with the SD card recovery method and verify the camera works without wyze hack first.

PDXUser, 2021-03-06:
I downgraded to what they have on the Wyze site which is 4.10.6.218, (and tried the hack again)  it says it completed audibly on the camera, but it doesn't seem like it is as the virtual SD Card is still not available which i have the NFS share mapped to in the config file.   I can run the Wyze firmware with no problems and record to SD Card, but I'd rather not, or record to their cloud.  

When on this hack, the camera still wants to reboot even with this firmware on a regular basis (every few minutes).  Without the firmware it doesn't.  The config file was only changed to have my NFS mapping so there was nothing else changed in the file.

HclX, 2021-03-06:
Is it possible your NFS server is not reachable?

PDXUser, 2021-03-06:
You replied as I was adding my edit to the comment for the install.log so I am not sure if you saw that edit, but it is there now if you want to peruse it for any info related to this.   As far as the NFS server not being visible, it is pingable from inside my network and I am connected to a share watching the Camera directory for any activity.

PDXUser, 2021-03-06:
I think the issue is not the firmware on the camera.   I traced the issue back to the last timestamp in my Camera directory which was from Feb 20th 2021.   On Feb 22 2021 the app updated to version 2.17.41 on my phone which I believe is the issue with the SD Card emulation not working.  Not so sure on the reboot problem though, I went through one more factory reset and this time, left it for an hour on stock with no reboots happening.  Back on this hack and it is having the reboot problem again.  I installed the RTSP version of the firmware which brought it down to 4.29.4.49  which I know for sure worked. 

I can't help but wonder if some of their "bug fixes' in the app are related to preventing this from happening.

HclX, 2021-03-13:
I have uploaded release 0.5.4 in `bugfixes` branch, please give it a try.

PDXUser, 2021-03-13:
I will test it for random reboots and let you know.  The SD Card emulation is working properly now though.  :+1:

HclX, 2021-03-13:
I had one pan camera went through the update not in an easy way, but
eventually it did go through. So I guess it's just individual problem.

PDXUser, 2021-03-13:
No more random reboots!  Good to close your ticket!

closed 2021-03-14 by HclX