HclX/WyzeHacks #98: New Firmware Version 4.9.6.241 seems to disable telnet and NFS

Semag, 2021-03-11:
Hey @HclX  - 

I thought i'd test upgrading one of my cameras to check for everyone and it seems that the new firmware released today disables telnet as well as the NFS.  I had not gotten the auto update feature working in my cameras prior to this so i'm not sure if the camera makes an attempt to do the auto-reinstall of the hack after upgrade, but i know for sure i had uploads from this camera early this morning and after upgrade I no longer have the NFS mount and also don't have the telnet available.

Just a warning if anyone loves the NFS as much as i do.

Semag, 2021-03-11:
False Alarm - i had to reinstall the latest version and it works now, my apologies.

Semag, 2021-03-11:
Actually - scratch that, i have my camera upgraded, reinstalled @HclX 's fix, and am not able to see the SD card in the app. It looks like it mounted the NFS correctly but it's not treating it as an SD card: I feel like these mount points are new too?

/dev/mtdblock3 on /driver type squashfs (ro,relatime)
/dev/mtdblock4 on /system type jffs2 (rw,relatime)
/dev/mtdblock7 on /backupa type jffs2 (rw,relatime)
/dev/mtdblock8 on /configs type jffs2 (rw,relatime)
/dev/mtdblock9 on /params type jffs2 (rw,relatime)

sensedesigns, 2021-03-12:
I am trying to install this one Wyze Cam v2 Firmware 4.9.6.241 and I seem to be having the same issue using remote_install.sh. I do not hear "installation begins" when I push the update and the command line seems to continually clock with ".............."

I pulled the SD card from the camera I was installing this on there is no install.log either.

Semag, 2021-03-12:
@sensedesigns  - did you have it previously installed?  I got the install to work fine with remote_install.sh and telnet is re-enabled on my upgraded camera.  However, the SD Emulation doesn't seem to be working.  I'm wondering if they combined a software branch from V3 based on the latest fix in the hack - and if that needs to get ported over to the v2's.

sensedesigns, 2021-03-12:
@Semag No, I didn't have it previously installed. I tried to install on firmware of 4.9.4.XXX and ran into the exact same issue as well.

HclX, 2021-03-12:
Sometimes firewall or strange networking settings can cause this. I
remember I was using vscode remote and it tried to redirect the http
service to my local machine, causing it not working...

Semag, 2021-03-12:
@HclX  - i didn't try to install an SD card to see if the issue was the same as the one https://github.com/HclX/WyzeHacks/issues/87 where on the v3 he had to install the SD card to get it working. I might try that today.

Semag, 2021-03-13:
One more piece - i tried putting in an SD Card - it looks like it tried to mount the virtual card (NFS) after but it didn't like it:

[14950.476874] jzmmc_v1.2 jzmmc_v1.2.0: card inserted, state=0
[14951.807959] mmc0: new high speed SDHC card at address 1234
[14951.816994] mmcblk0: mmc0:1234 SA16G 14.4 GiB
[14951.824172]  mmcblk0: p1
[15357.033497] jzmmc_v1.2 jzmmc_v1.2.0: card removed, state=0
[15357.039498] mmc0: card 1234 removed
[15357.089117] FAT-fs (mmcblk0p1): FAT read failed (blocknr 928)
[15357.116625] FAT-fs (mmcblk0p1): FAT read failed (blocknr 928)
[15357.133171] FAT-fs (mmcblk0p1): FAT read failed (blocknr 928)
[15357.156274] FAT-fs (mmcblk0p1): FAT read failed (blocknr 927)
[15357.264397] FAT-fs (mmcblk0p1): unable to read boot sector to mark fs as dirty
[15357.279428] FAT-fs (mmcblk0p1): bread failed in fat_clusters_flush

HclX, 2021-03-13:
I uploaded a new release in `bugfixes` branch.  Please read the release note and test it for me.

Semag, 2021-03-13:
@HclX  Awesome buddy, that bugfix branch fixed it!

HclX, 2021-03-14:
released in master branch now, closing this issue.

closed 2021-03-14 by HclX