HclX/WyzeHacks #99: Is WyzeHacks expected to work with v1 cameras?

jamesderlin, 2021-03-15:
It's unclear from the `README.md` whether WyzeHacks is expected to work for v1 cameras; "v1" is mentioned only for the uninstall instructions.

I tried using the SD card install approach, and I heard the "Upgrade is in progress" message, but I never heard the "Installation finished" message.  There also was no `install.log` file left on the SD card.

HclX, 2021-03-15:
The answer is "it might", reason is simple: I don't have one to test. Also, te SD card install method no longer works due to recent firmware changes, you should instead use the remote_install.sh. and again, I have no way of testing I'd it works on v1.

jamesderlin, 2021-03-15:
Ah, ok.  I thought that the SD card install method would work since the v1 firmware hasn't been updated in a while.

I've now tried `remote_install.sh`, and it appears to skip over my v1 cameras. =(

(I'd be happy to donate a v1 camera if you're interested.)

HclX, 2021-03-16:
I see. I can take a look if you can mail me one of the v1 device, but no
promise how soon I can get it work.

HclX, 2021-04-16:
got the v1 camera, and noticed this is on a totally different architecture: v2 and v3 are all using Ingenic's SoC, which is MIPS based, while v1 seems to be ARM based. While most of the hack is just shell scripts, there is a critical component built with C, which means I need to get the toolchains and SDK for that. I will definitely look into it, but I highly suggest you migrate your system to v2/v3 if you want to use this hack.