HclX/WyzeHacks #100: Unable to install WyzeHacks on V3 Camera

barthohomeautomation, 2021-03-17:
I have been attempting to install wyze_hacks_0_5_04 on some v3 cameras I received. I installed the 4.36.0.228 firmware found within the tools/v3/firmwares folder. I was unable to find init_v3.sh as directed int the directions but I did find v3_init.sh. I ran this and then ran remote_install.sh. There were to voice prompts as seen in v2 of the camera but all seem's to have installed. Unfortunately, after install, I cannot telnet into the camera, and remote storage is not enabled. Any thoughts on what I may be doing wrong?

sensedesigns, 2021-03-17:
I have the same issues, here's the steps to reproduce:

- WyzeCam v3
- firmware 4.36.0.88
- Trying to install via command line on chromebook
- NFS mount is SMB 1
- Run ./remote_install.sh

Output:

```
Found local config file, including into firmware update archive...
Upgrade/config.inc
Requirement already satisfied: requests in /opt/miniconda3/lib/python3.8/site-packages (2.24.0)
Requirement already satisfied: chardet<4,>=3.0.2 in /opt/miniconda3/lib/python3.8/site-packages (from requests) (3.0.4)
Requirement already satisfied: urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 in /opt/miniconda3/lib/python3.8/site-packages (from requests) (1.25.11)
Requirement already satisfied: certifi>=2017.4.17 in /opt/miniconda3/lib/python3.8/site-packages (from requests) (2020.6.20)
Requirement already satisfied: idna<3,>=2.5 in /opt/miniconda3/lib/python3.8/site-packages (from requests) (2.10)
INFO:root:Trying saved credentials from /home/gregairel/.wyze_token.
INFO:root:Checking device, mac=7C78B20C54DD

Device type:      Camera (WYZE_CAKP2JFUS)
Device name:      Camv3test
Firmware version: 4.36.0.88
IP Address:       192.168.1.9

Pushing firmware to this device? [y/N]:y
INFO:root:Serving firmware file './firmware.bin' as 'http://100.115.92.199:11808/firmware.bin', md5=46fa9ae2b27081fe386fd2148e8842a4
INFO:root:Checking device, mac=2CAA8EA50415

Device type:      Camera (WYZEC1-JZ)
Device name:       Test
Firmware version: 4.9.4.175
IP Address:       192.168.1.6

Pushing firmware to this device? [y/N]:
```

- The light turns to purple blinking on the camera after I push the firmware.
- The output of the command line just continues to output "......"
- I never hear "installation begins"

HclX, 2021-03-18:
Sound play not wot on v3 is a known issue. But other than that it should work. Your case it seems the update is not even pushed to camera, you should see some message when it is happening.

barthohomeautomation, 2021-03-18:
I receive a line stating "GET /firmware.bin HTTP/1.1" 200 -". Once this happens the LED on the camera starts flashing purple for a few minutes and then reboots. This is when running v3.init_sh. When running remote_install.sh, I receive the same "GET /firmware.bin HTTP/1.1" 200 -" line but the camera reboots after only a few seconds. I have tried this with two cameras.

barthohomeautomation, 2021-03-18:
I was able to resolve the issue. Today, I attempted the process again a total of 5 times. On the 5th time, I was successful in the install. The only difference was I left the SD card in for the entire process on the 5th attempt. Previously I would remove the CD card after installing the 4.36.0.228 firmware. Then once the SD card is removed I would complete the process of running v3_init.sh and remote_install.sh. The fifth time I simply forgot to remove the card and went through the process. It worked. I tried to replicate it on the second camera and I was able to. If I removed the SD card after installing 4.36.0.228, the process would complete and it appeared all worked as it should. But I was not able to telnet into the cameras and it was not saving files remotely. But if I completed the process with the SD card in, and removed it after the process, all seems to work. Not sure if this is something unique to my implementation but all seem good now.

HclX, 2021-03-18:
That is interesting. So to clarify:
The hack installation requires the presence of physical sd card, however,
after installation, the hack works without physical sd card. Is that
correct?

Can you reboot the camera and verify if it's still working without sd card?

barthohomeautomation, 2021-03-18:
Yes, that is correct. I was not able to get the hack to install if I removed the SD card after the manual firmware install, but before installing the hack. But if I left the card in for the whole process, and removed the card after the hack install, I was able to telnet and the cameras are recording remotely. I have confirmed they are actually recording as I have viewed the recorded files stored on the remote storage.

HclX, 2021-03-18:
I think I know why: can you check if you have a config.inc file on sdcard?
You will need one either on sdcard or at the installer directory.

barthohomeautomation, 2021-03-18:
I just checked, there is no config file on the SD card. Just the firmware .bin file along with the files/directories created from the install and few seconds of recording. But there is one in the install directory on the source computer. This is the config I edited for my use case.

zarthan, 2021-03-20:
I have experienced the unable to install. I ran the remote_install the first time without copying the config file. I hid have telnet. I edited copied and edited the config and redid the install. The second time, I no longer had telnet access and as reported earlier, the firmware update stops and the camera resets after a few seconds. I have tried with an SD card and without. I rolled back to the original firmware, reset the camera. I have run the script dozens of times and occasionally the firmware update process does take several minutes as it should, but I still do not have telnet nor does the SD card show up in the app.

barthohomeautomation, 2021-03-20:
> I have experienced the unable to install. I ran the remote_install the first time without copying the config file. I hid have telnet. I edited copied and edited the config and redid the install. The [...]

You may want to create a separate issue to help keep issues organized for the developer.

zarthan, 2021-03-20:
Looks like the same issue to me.

HclX, 2021-03-27:
For those running into issues, I have a recent change pushed to bugfixes branch. This change does nothing but integrates the init process into installation so all you need is running remote install and it should work. Please give it a try: if it doesn't work, please put an sd card in camera while running the installer and paste install.log here.

zarthan, 2021-03-27:
The install still fails with the yesterdays update. I don't seem to get an install.log file with the 0.5.05 
The actual wyze_updater.py was not in the zip file. It looks like the symlink was not followed, so just the link no actual file. I copied wyze_updater.py from the previous version so that may account for the numbering error. The install log below is what I had on the SD card when I pulled it out after running the installer, but the DS card had been in during all my previous install runs with  0.5.04. I copied all the files off and reformatted the SD card. Restarted the camera and inserted the SD card. Re-ran the remote_install script. Pulled the SD card and I did not find the install.log this time. I ran the remote_install script several times but no install.log and no telnet access.
Let me know if I should try something else.

Log output from 0.5.04
+ echo WyzeHack: Starting wyze hack installer...
WyzeHack: Starting wyze hack installer...
+ play_sound /tmp/Upgrade/install_snd/begin.wav 50
/tmp/wyze_hack/install/main.sh: line 1: can't create /sys/class/gpio/gpio63/value: nonexistent directory
+ echo 1
+ /tmp/wyze_hack/install/bin/audioplay /tmp/Upgrade/install_snd/begin.wav 50
/tmp/wyze_hack/install/main.sh: line 1: can't create /sys/class/gpio/gpio63/value: nonexistent directory
+ echo 0
+ [ -f /media/mmc/debug/.copyfiles ]
+ pgrep -f telnetd
+ set_passwd root::10933:0:99999:7:::
+ /bin/umount /etc
umount: can't umount /etc: Invalid argument
+ rm -rf /tmp/etc
+ cp -r /etc /tmp/
+ echo root::10933:0:99999:7:::
+ /bin/mount -o bind /tmp/etc /etc
+ echo WyzeHack: Enabling telnetd...
WyzeHack: Enabling telnetd...
+ busybox telnetd
+ [ -z 4.36.0.252 ]
+ echo WyzeHack: Current Wyze software version is 4.36.0.252
WyzeHack: Current Wyze software version is 4.36.0.252
+ echo WyzeHack: Installing WyzeHacks version 0.5.04
WyzeHack: Installing WyzeHacks version 0.5.04
+ [ -f /params/wyze_hack.cfg ]
+ [ -f /tmp/Upgrade/config.inc ]
+ [ -f /media/mmc/config.inc ]
+ [ ! -f /configs/wyze_hack.cfg ]
+ [ -z V3 ]
+ echo WyzeHack: Copying wyze hack binary...
WyzeHack: Copying wyze hack binary...
+ cp /tmp/Upgrade/wyze_hack.sh /configs/wyze_hack.sh
+ echo WyzeHack: Hooking up boot script...
WyzeHack: Hooking up boot script...
+ hook_init
+ grep wyze_hack.sh /etc/init.d/rcS
+ [ ! -f /configs/wyze_hack.sh ]
+ [ ! -f /configs/wyze_hack.cfg ]
+ local SYSTEM_DIR=/system
+ [ ! -L /system/init/app_init.sh ]
+ echo WyzeHack: backing up app_init.sh ...
WyzeHack: backing up app_init.sh ...
+ cp /system/init/app_init.sh /system/init/app_init_orig.sh
cp: can't create '/system/init/app_init_orig.sh': Read-only file system
+ readlink /system/init/app_init.sh
+ local APP_INIT=
+ [  != /configs/wyze_hack.sh ]
+ echo WyzeHack: setting up symlink ...
WyzeHack: setting up symlink ...
+ ln -s -f /configs/wyze_hack.sh /system/init/app_init.sh
ln: /system/init/app_init.sh: File exists
+ ls -la /system/init
drwxrwxr-x    2 1000     1000           104 Feb  3 01:00 .
drwxrwxr-x    8 1000     1000            87 Feb  3 01:00 ..
-rwxrwxr-x    1 1000     1000          2207 Feb  2 22:05 app_init.sh
-rwxrwxr-x    1 1000     1000           148 Dec 30 06:07 factory.sh
-rw-r--r--    1 1000     1000            11 Feb  3 01:00 firmware_build_epoch_time.txt
-rwxrwxr-x    1 1000     1000           574 Nov 11 03:57 wifi.sh
+ return 0
+ play_sound /tmp/Upgrade/install_snd/finished.wav 50
/tmp/wyze_hack/install/main.sh: line 1: can't create /sys/class/gpio/gpio63/value: nonexistent directory
+ echo 1
+ /tmp/wyze_hack/install/bin/audioplay /tmp/Upgrade/install_snd/finished.wav 50
/tmp/wyze_hack/install/main.sh: line 1: can't create /sys/class/gpio/gpio63/value: nonexistent directory
+ echo 0
+ rm /media/mmc/version.ini.old
+ mv /media/mmc/version.ini /media/mmc/version.ini.old
+ /sbin/reboot

HclX, 2021-03-27:
@zarthan your log shows you either didn't run "v3_init.sh" prior the installation of 0.5.04, or the init thing failed for some reason. There might be an init.log on sdcard for details.

Anyway, v3_init.sh is no longer needed with 0.5.05. I don't know why it didn't work and didn't create logs though.

zarthan, 2021-03-27:
I did run v3_init although I did not see anything in the readme only in this issue thread.
I re-ran 0.5.0.4 with an SD card. 
the contents of the v3_init.log
Installing...
Enabling telnetd...
Incorrect app version: 4.36.0.252

I imagine the install log will be the same but here for completeness

+ echo WyzeHack: Starting wyze hack installer...
WyzeHack: Starting wyze hack installer...
+ play_sound /tmp/Upgrade/install_snd/begin.wav 50
/tmp/wyze_hack/install/main.sh: line 1: can't create /sys/class/gpio/gpio63/value: nonexistent directory
+ echo 1
+ /tmp/wyze_hack/install/bin/audioplay /tmp/Upgrade/install_snd/begin.wav 50
/tmp/wyze_hack/install/main.sh: line 1: can't create /sys/class/gpio/gpio63/value: nonexistent directory
+ echo 0
+ [ -f /media/mmc/debug/.copyfiles ]
+ pgrep -f telnetd
+ set_passwd root::10933:0:99999:7:::
+ /bin/umount /etc
umount: can't umount /etc: Invalid argument
+ rm -rf /tmp/etc
+ cp -r /etc /tmp/
+ echo root::10933:0:99999:7:::
+ /bin/mount -o bind /tmp/etc /etc
+ echo WyzeHack: Enabling telnetd...
WyzeHack: Enabling telnetd...
+ busybox telnetd
+ [ -z 4.36.0.252 ]
+ echo WyzeHack: Current Wyze software version is 4.36.0.252
WyzeHack: Current Wyze software version is 4.36.0.252
+ echo WyzeHack: Installing WyzeHacks version 0.5.04
WyzeHack: Installing WyzeHacks version 0.5.04
+ [ -f /params/wyze_hack.cfg ]
+ [ -f /tmp/Upgrade/config.inc ]
+ [ -f /media/mmc/config.inc ]
+ [ ! -f /configs/wyze_hack.cfg ]
+ [ -z V3 ]
+ echo WyzeHack: Copying wyze hack binary...
WyzeHack: Copying wyze hack binary...
+ cp /tmp/Upgrade/wyze_hack.sh /configs/wyze_hack.sh
+ echo WyzeHack: Hooking up boot script...
WyzeHack: Hooking up boot script...
+ hook_init
+ grep wyze_hack.sh /etc/init.d/rcS
+ [ ! -f /configs/wyze_hack.sh ]
+ [ ! -f /configs/wyze_hack.cfg ]
+ local SYSTEM_DIR=/system
+ [ ! -L /system/init/app_init.sh ]
+ echo WyzeHack: backing up app_init.sh ...
WyzeHack: backing up app_init.sh ...
+ cp /system/init/app_init.sh /system/init/app_init_orig.sh
cp: can't create '/system/init/app_init_orig.sh': Read-only file system
+ readlink /system/init/app_init.sh
+ local APP_INIT=
+ [  != /configs/wyze_hack.sh ]
+ echo WyzeHack: setting up symlink ...
WyzeHack: setting up symlink ...
+ ln -s -f /configs/wyze_hack.sh /system/init/app_init.sh
ln: /system/init/app_init.sh: File exists
+ ls -la /system/init
drwxrwxr-x    2 1000     1000           104 Feb  3 01:00 .
drwxrwxr-x    8 1000     1000            87 Feb  3 01:00 ..
-rwxrwxr-x    1 1000     1000          2207 Feb  2 22:05 app_init.sh
-rwxrwxr-x    1 1000     1000           148 Dec 30 06:07 factory.sh
-rw-r--r--    1 1000     1000            11 Feb  3 01:00 firmware_build_epoch_time.txt
-rwxrwxr-x    1 1000     1000           574 Nov 11 03:57 wifi.sh
+ return 0
+ play_sound /tmp/Upgrade/install_snd/finished.wav 50
/tmp/wyze_hack/install/main.sh: line 1: can't create /sys/class/gpio/gpio63/value: nonexistent directory
+ echo 1
+ /tmp/wyze_hack/install/bin/audioplay /tmp/Upgrade/install_snd/finished.wav 50
/tmp/wyze_hack/install/main.sh: line 1: can't create /sys/class/gpio/gpio63/value: nonexistent directory
+ echo 0
+ rm /media/mmc/version.ini.old
+ mv /media/mmc/version.ini /media/mmc/version.ini.old
+ /sbin/reboot

zarthan, 2021-03-27:
When I saw the "Incorrect app version: 4.36.0.252" from the v3_init log, I decided to downgrade the firmware to 4.36.0.228 I re-ran v3_init and got telnet. I then ran remote_install and now have a successful install. I didn't look at the wyze_updater to see whether there was something that checked the firmware version. 

I used the 0.5.0.4 version of the script. the 0.5.0.5 version did not have the wyze_updater.py file, just a symlink to didn't point to anything. Also, the script complained about the firmware_660 bin file being missing and it is.

zarthan, 2021-03-27:
I am going to leave the firmware at the current level for a little bit because I need to do some testing work. 

What is the V3 camera root telnet login password?

HclX, 2021-03-27:
The builtin default password is WYom2020

If you have uncommented the default password shadow in config file then
it's changed to ismart12

zarthan, 2021-03-27:
Many thanks. 
Once I have done some testing here, I will re-flash with older firmware again and then run the newest script again

HclX, 2021-03-27:
the firmware version restriction is removed in 0.5.5 so you should be able
to do a fresh install on any firmware.

musicjeremy, 2021-04-10:
I am also having a problem - here is the log from the sdcard after doing a remote install to a cam v3:

+ echo WyzeHack: Starting wyze hack installer...
WyzeHack: Starting wyze hack installer...
+ play_sound /tmp/Upgrade/install_snd/begin.wav 50
/tmp/wyze_hack/install/main.sh: line 1: can't create /sys/class/gpio/gpio63/value: nonexistent directory
+ echo 1
+ /tmp/wyze_hack/install/bin/audioplay /tmp/Upgrade/install_snd/begin.wav 50
/tmp/wyze_hack/install/main.sh: line 1: can't create /sys/class/gpio/gpio63/value: nonexistent directory
+ echo 0
+ [ -f /media/mmc/debug/.copyfiles ]
+ pgrep -f telnetd
+ set_passwd root::10933:0:99999:7:::
+ /bin/umount /etc
umount: can't umount /etc: Invalid argument
+ rm -rf /tmp/etc
+ cp -r /etc /tmp/
+ echo root::10933:0:99999:7:::
+ /bin/mount -o bind /tmp/etc /etc
+ echo WyzeHack: Enabling telnetd...
WyzeHack: Enabling telnetd...
+ busybox telnetd
+ [ -z 4.36.0.280 ]
+ echo WyzeHack: Current Wyze software version is 4.36.0.280
WyzeHack: Current Wyze software version is 4.36.0.280
+ echo WyzeHack: Installing WyzeHacks version 0.5.05
WyzeHack: Installing WyzeHacks version 0.5.05
+ [ -f /params/wyze_hack.cfg ]
+ [ -f /tmp/Upgrade/config.inc ]
+ echo WyzeHack: Use config file /tmp/Upgrade/config.inc
WyzeHack: Use config file /tmp/Upgrade/config.inc
+ sed s/\r$// /tmp/Upgrade/config.inc
+ [ -f /media/mmc/config.inc ]
+ [ ! -f /configs/wyze_hack.cfg ]
+ [ -z V3 ]
+ echo WyzeHack: Copying wyze hack binary...
WyzeHack: Copying wyze hack binary...
+ cp /tmp/Upgrade/wyze_hack.sh /configs/wyze_hack.sh
+ init_rootfs /tmp/Upgrade/rootfs
+ grep wyze_hack.sh /etc/init.d/rcS
+ [ V3 != V3 ]
+ grep -i AppVer /usr/app.ver
+ ROOTFS_VER=4.36.0.56
+ ROOTFS_BIN=/tmp/Upgrade/rootfs/4.36.0.56.bin
+ [ ! -f /tmp/Upgrade/rootfs/4.36.0.56.bin ]
+ echo WyzeHack: No root fs image found for version 4.36.0.56
WyzeHack: No root fs image found for version 4.36.0.56
+ return 1
+ echo WyzeHack: Init root fs failed, aborting...
WyzeHack: Init root fs failed, aborting...
+ play_sound /tmp/Upgrade/install_snd/failed.wav 50
/tmp/wyze_hack/install/main.sh: line 1: can't create /sys/class/gpio/gpio63/value: nonexistent directory
+ echo 1
+ /tmp/wyze_hack/install/bin/audioplay /tmp/Upgrade/install_snd/failed.wav 50
/tmp/wyze_hack/install/main.sh: line 1: can't create /sys/class/gpio/gpio63/value: nonexistent directory
+ echo 0
+ return 1

lilchancep, 2021-04-12:
`+ echo WyzeHack: Starting wyze hack installer...
WyzeHack: Starting wyze hack installer...
+ play_sound /tmp/Upgrade/install_snd/begin.wav 50
/tmp/wyze_hack/install/main.sh: line 1: can't create /sys/class/gpio/gpio63/value: nonexistent directory
+ echo 1
+ /tmp/wyze_hack/install/bin/audioplay /tmp/Upgrade/install_snd/begin.wav 50
/tmp/wyze_hack/install/main.sh: line 1: can't create /sys/class/gpio/gpio63/value: nonexistent directory
+ echo 0
+ [ -f /media/mmc/debug/.copyfiles ]
+ pgrep -f telnetd
+ set_passwd root::10933:0:99999:7:::
+ /bin/umount /etc
umount: can't umount /etc: Invalid argument
+ rm -rf /tmp/etc
+ cp -r /etc /tmp/
+ echo root::10933:0:99999:7:::
+ /bin/mount -o bind /tmp/etc /etc
+ echo WyzeHack: Enabling telnetd...
WyzeHack: Enabling telnetd...
+ busybox telnetd
+ [ -z 4.36.0.280 ]
+ echo WyzeHack: Current Wyze software version is 4.36.0.280
WyzeHack: Current Wyze software version is 4.36.0.280
+ echo WyzeHack: Installing WyzeHacks version 0.5.05
WyzeHack: Installing WyzeHacks version 0.5.05
+ [ -f /params/wyze_hack.cfg ]
+ [ -f /tmp/Upgrade/config.inc ]
+ echo WyzeHack: Use config file /tmp/Upgrade/config.inc
WyzeHack: Use config file /tmp/Upgrade/config.inc
+ sed s/\r$// /tmp/Upgrade/config.inc
+ [ -f /media/mmc/config.inc ]
+ [ ! -f /configs/wyze_hack.cfg ]
+ [ -z V3 ]
+ echo WyzeHack: Copying wyze hack binary...
WyzeHack: Copying wyze hack binary...
+ cp /tmp/Upgrade/wyze_hack.sh /configs/wyze_hack.sh
+ init_rootfs /tmp/Upgrade/rootfs
+ grep wyze_hack.sh /etc/init.d/rcS
+ [ V3 != V3 ]
+ sed -E s/^.*=[[:space:]]*([0-9.]+)[[:space:]]*$/\1/g
+ grep -i AppVer /usr/app.ver
+ ROOTFS_VER=4.36.0.56
+ ROOTFS_BIN=/tmp/Upgrade/rootfs/4.36.0.56.bin
+ [ ! -f /tmp/Upgrade/rootfs/4.36.0.56.bin ]
+ echo WyzeHack: No root fs image found for version 4.36.0.56
WyzeHack: No root fs image found for version 4.36.0.56
+ return 1
+ echo WyzeHack: Init root fs failed, aborting...
WyzeHack: Init root fs failed, aborting...
+ play_sound /tmp/Upgrade/install_snd/failed.wav 50
/tmp/wyze_hack/install/main.sh: line 1: can't create /sys/class/gpio/gpio63/value: nonexistent directory
+ echo 1
+ /tmp/wyze_hack/install/bin/audioplay /tmp/Upgrade/install_snd/failed.wav 50
/tmp/wyze_hack/install/main.sh: line 1: can't create /sys/class/gpio/gpio63/value: nonexistent directory
+ echo 0
+ return 1
`

Here in my installation log for my WyzeCam V3 that i could not get working.

HclX, 2021-04-12:
@musicjeremy @lilchancep To get around this issue, please use SD card recovery method to reflash your camera to firmware 4.36.0.280 and try it again.

Reason behind this error:
To get the hack work on v3 camera, I have to modify the rootfs image. The first installation will basically re-flash your camera with a modified rootfs image. This rootfs modification can only be done on desktop, so it has to be based on a specific firmware version. When I was doing that I took the image from firmware 4.36.0.218. To avoid version inconsistencies I added a check to make sure the rootfs on the camera matches the one I found from firmware 4.36.0.218.  If it's not, the installation will fail.

musicjeremy, 2021-04-14:
Thank you @HclX - just to confirm:

1. Your script checks for 4.36.0.2**18**
2. We should flash with 4.36.0.2**80** and try again?

> @musicjeremy @lilchancep To get around this issue, please use SD card recovery method to reflash your camera to firmware 4.36.0.280 and try it again.
>
> Reason behind this error:
> To get the hack work [...]

sveip, 2021-04-14:
Maybe it's easier to build a few rootfs variants for different versions and
make the installer choose the appropriate one?

nijhawank, 2021-04-14:
So your release includes the rootfs from 4.36.0.218 but when I extracted your latest zip and the FIRMWARE_660R.bin inside, I see that the rootfs is 4.36.0.112. Which one is right, 112 or 218?
Does that mean flashing the WyzeHacks on my cam that is on 4.36.0.280 will be rolled back to 4.36.0.112-or-218?

HclX, 2021-04-14:
There are multiple images in demo.bin. The versions app shows or you see on
wyze release note is app image version. The rootfs image version won't
change during regular ota update (as far as I know), it only changes when
you recover using sd card method. Wyzehack installer will only install if
it finds your rootfs version matches the expected version, so it won't
downgrade or upgrade your rootfs version.

lilchancep, 2021-04-16:
@HclX Would it be possible to get a version 0.5.05 on the releases page?

HclX, 2021-04-16:
@lilchancep ah, totally forgot that, I updated the release page.

lilchancep, 2021-04-17:
@HclX Do you have a link to the firmware i should be using to flash my V3 camera. as version 5.05 is still not working.

Again thank you so much

HclX, 2021-04-18:
280 should work.

lilchancep, 2021-04-18:
@HclX 
1. Flashed wyze cam with V3 firmware 4.36.0.280

2. Downloaded and extracted v 0.5.05

3. copied and set up config.inc
	Line Changes
	export NFS_ROOT='10.2.0.11:/mnt/MainPool/ApplicationData/Wyze'
	export NFS_OPTIONS='-o nolock,rw,noatime,nodiratime'
        export SYNC_BOOT_LOG=1

4. ran remote_intall.sh
	Camera has purple light
	Reboots in about a minute or so

The camera starts to reboot every 15 seconds (i think this is due to lack of NFS share access.

It makes the folder and writes on my NAS just fine and writes files. just not recording and the camera says "NO SD Card Detected"

I can provide any files that you think will be helpful.

HclX, 2021-04-19:
Try telnet into the camera and check /tmp/boot.log? Make sure you filter
the content before posting it here as it contains sensitive information.

HclX, 2021-04-25:
I'm closing this issue as I don't see any update for some days.

closed 2021-04-25 by HclX