HclX/WyzeHacks #101: Directory permission issue with firmware 4.9.6.241

unomar, 2021-03-18:
I'm curious if others are seeing this, but since upgrading my V2's to 4.9.6.241 any new directories created under the record dir have 644 permissions instead of 755.  This then prevents the creation of anything below it as the camera doesn't have the ability to execute on directories.  I tried re-flashing the stock firmware and re-applying WyzeHacks, but no luck.  Reverting to 4.9.6.199 restored the functionality (directories 755, files 644).

unomar, 2021-03-19:
Interesting.  Out of curiosity I flashed one of my cams to stock 6.241 and let it record to the actual SD card.  That worked as expected creating directories with the correct permissions.  So some combination of that firmware with this mod and my NFS server just isn't happy.  

Previously I had been running wyzehacks v 0.4.04 and everything worked great until the 6.241 firmware update which forced me to upgrade my wyzehacks to 0.5.04.

I'll look into NFS user mapping, but I might also dig around in the code to see if I can figure out why it suddenly broke.

HclX, 2021-03-19:
The lastest version hack should also work with older version firmware. Can you try different combinations to isolate the issue?

unomar, 2021-03-19:
Permissions are correct with 0.5.04 and 4.9.6.199.

I'll test 4.9.6.218 tomorrow.  Also, I just noticed that 6.241 has this note on the download page
`This firmware update has been paused while we investigate the cause of an issue that may be related to this update. If we find that the update is not the cause, we will resume the rollout.`

HclX, 2021-03-19:
Sounds like they messed up again. Also, I believe if it's physical sdcard,
it is formated as fat which doesn't support linux permissions, which is why
it's working with real sd card.

It's also worth mentioning all my cameras are running version 241 with
latest hack and I don't have this issue. So it might be a combination of
firmware and NFS settings.

unomar, 2021-03-19:
Good call on the FAT32 permissions.  I didn't even consider that.  It's unfortunate as it means they might not care to fix it since it wouldn't affect normal SD card users.  The chatter on the Wyze Forums seems to indicate the reason the firmware rollout was paused is due to sound sensitivity which doesn't seem related to my issue.

Permissions are correct with 0.5.04 and 4.9.6.218.

So it does appear to be something injected in the 241 firmware that is incompatible with my NFS settings.

unomar, 2021-03-22:
Closing as this appears to be caused by Wyze firmware & local NFS configuration.

closed 2021-03-22 by unomar

ChrisMacNaughton, 2021-03-27:
I'm seeing exactly this issue using NFSv3 (`rw,sync,no_subtree_check,all_squash,anonuid=1000,anongid=1000`) on a ZFS backed NFS. I'm also using the latest (4.9.6.241) firmware and am thus, without changing directory permissions manually, unable to save recordings to my server.

ChrisMacNaughton, 2021-03-27:
Downgrading to 4.9.6.218 is enough to resolve the permission issues.