HclX/WyzeHacks #102: Toolchain for wyze cameras.

cburns4563, 2021-03-19:
Hello,

Long time lurker here. I’m not using this project but I’m interested in messing about with these devices. I was wondering if I could get a copy of your compiler/toolchain. I have one based on musl but it doesn’t support dynamic loading. I seem to recall seeing a link to the toolchain on GitHub somewhere but now I can’t find it.

Thanks for your consideration and great project!

Charlie

cburns4563, 2021-03-19:
I found one and am good to go. Thanks.

closed 2021-03-19 by cburns4563

endertable, 2021-03-20:
> 
> 
> I found one and am good to go. Thanks.

HI @cburns4563 I have also been seeking a correct toolchain for the cam.  I guess my first question to you is are you going to compile programs for the stock cam firmware or are you going to create a whole new build/system?  If you are creating for stock, can I ask where you got your toolchain.  I have one (mips-gcc472-glibc216)  but it is glibc and need one for uClibc which is what cam uses.

Thanks

cburns4563, 2021-03-20:
I’m just writing small programs to run on the camera. 

Building toolchains from scratch is above my pay grade so I googled around for a long time looking for an easier approach.

This is what worked for me:

https://sourcery.mentor.com/GNUToolchain/release3217

The documentation is quite good:

https://sourcery.mentor.com/GNUToolchain/doc23862/getting-started.pdf

Good luck and have fun.

Charlie