HclX/WyzeHacks #103: NFS is not recognized as SD with 4.9.6.241

vrurg, 2021-03-22:
I just have realized today that my cameras do not record on NFS due to the updated firmware. So, updated the `master` and re-installed. Got telnet and NFS back but the camera still doesn't consider it as SD.

Here is my config, the same used on all cameras. Apparently, some sensitive info removed.

```
# USER CONFIGURATIONS -- TO BE MODIFIED BY USER
# Make your own copy of this file and put it under the root of SD card
# with name "config.inc" when installing the hack.

# You need to modify NFS_ROOT to point to your NFS share. The recordings
# will be stored under $NFS_ROOT/WyzeCams/<MAC_ADDR_OF_YOUR_CAMERA>.
export NFS_ROOT='<server>:/Recordings'

# Some NFS configurations may need special mount options. You can specify
# them here. The default value is verified on my camera with a NFS v3 share
# on my home server.
# export NFS_OPTIONS='-o nolock,rw,noatime,nodiratime'

# Timeout threshold for NFS connectivity check. If the NFS share is not
# available for more than '$NFS_TIMEOUT' seconds, a reboot will be initiated
# trying to resolve the network connectivity. The default value is 15 seconds.
# export NFS_TIMEOUT=15

# Uncomment and modify the AUTO_REBOOT variable to automatically reboot
# the camera at a specific minute. Time is specified in "HH:MM" format
# in local time.
# export REBOOT_AT='13:16'

# Uncomment and modify the ARCHIVE_OLDER_THAN variable to automatically
# archive recording clips and alarm images if it's a certain days older.
export ARCHIVE_OLDER_THAN=14

# Uncomment the following variable to sync the console log to the camera
# folder in NFS share. Use with caution as the log contains sensitive
# information and will cause your account compromise if leaked.
export SYNC_BOOT_LOG=1

# In my setup I noticed at some occasions the camera starts rebooting frequently
# which seems to be a result of dropping wifi connections. Running a "ping"
# command in background seems to be solving this issue. Uncomment this variable
# to enable this fix.
# export PING_KEEPALIVE=1

# Uncomment to enable voice notification when NFS network connection lost. The
# value should be in range of 0 and 100
# export NOTIFICATION_VOLUME=80

# Uncomment and update this value to customize the hostname of the camera. By
# default it will be set to <WyzeCam>-<last 4 hex digits of the mac>.
export HOSTNAME='cam-hallway'

# Uncomment this to enable automatically updating config.inc file based on file
# <camera_folder>/wyzehacks/config.new. This will be checked every one minute.
export AUTO_CONFIG=1

# Uncomment this to enable automatically updating wyzehacks. To update, you need
# to extract the latest wyzehacks release archive to $UPDATE_DIR and name it as
# "release_?_?_??". The updating script will search for latest version based on
# the folder name every one minute.
export AUTO_UPDATE=1

# By default, variable "UPDATE_DIR" is set to "/mnt/WyzeCams/wyzehacks", which
# is <NFS_ROOT>/WyzeCams/wyzehacks. If auto update is enabled, it will search
# for updates in this directory. The default value allows all cameras share the
# same update. You can customize this directory for a specific camera if you
# want it to use different update (for example, testing dev builds). Some thing
# under "/media/mmc" would be a good candidate because it maps to the camera
# folder.
# export UPDATE_DIR='/media/mmc/wyzehacks'

# Uncomment and update this to run a custom script after the NFS mount finishes.
# The script must already exists and executable. You can always use "/media/mmc"
# to refer the camera folder on the NFS share. Or use "/mnt" to refer the root
# of NFS share. The value of the variable can only contain script file name, no
# command line arguments. The script will be executed in background.
# export CUSTOM_SCRIPT='/media/mmc/scripts/myscript.sh'
# export CUSTOM_SCRIPT='/mnt/WyzeCams/scripts/myscript.sh'
```

With this config I do have NFS mounts alive:

```
/Recordings on /mnt type nfs
/Recordings/WyzeCams/XXXXXXXXXXXXXX on /media/mmcblk0p1 type nfs
/Recordings/WyzeCams/XXXXXXXXXXXXXX on /media/mmc type nfs
```

They're writable. There is no issue here. But what is curios is this excerpt from boot log:

```
WyzeHack: Mounting directory /mnt/WyzeCams/XXXXXXXXXXXX/ as SD card
WyzeHack: Mounting camera directory <server>:/Recordings//mnt/WyzeCams/XXXXXXXXXXXX/ on /media/mmcblk0p1
Not SD card, passing through...
WyzeHack: Mounting camera directory <server>:/Recordings//mnt/WyzeCams/XXXXXXXXXXXX/ on /media/mmc
Not SD card, passing through...
WyzeHack: Notifying iCamera about SD card insertion event...
sendto failed: Connection refused
WyzeHack: Custom script not found:
434
```

I think the problem should be hidden behind the  `sendto failed` line. I have traced it down to the location in the main.sh script and tried to reproduce it manually:

```
[root@cam-hallway:~]# /tmp/wyze_hack/run/bin/hackutils mmc_insert
sendto failed: Connection refused
```

Unfortunately, I don't have time to analyze it any further.

Semag, 2021-03-23:
Is that a V2 Camera @vrurg  ?  I Had the same issue in #98  but the updates that @HclX  added last week fixed it.

vrurg, 2021-03-23:
It is v2, yes. I saw the other issue. But it ends with fixes merged to master, and the master is what I use. So, I guess this time it's something different.

Semag, 2021-03-23:
@vrurg  - i never installed the master branch - you can try the "bugfixes" branch - that's the one i think i'm running.  Just in case?

vrurg, 2021-03-23:
Will try later today or tomorrow. Though I think I saw the merge in the git log.

vrurg, 2021-03-24:
As I expected, `bugfixes` branch is no different from `master`, `git diff` confirms it. Re-installation from `bugfixes` also didn't change the situation.

HclX, 2021-03-25:
Master has the latest change. This is strange as I tested on a couple of v2 cameras running on this version firmware. I will come up with some instructions for debugging tonight.

HclX, 2021-03-25:
@vrurg Here is what I would do to debug what's going on:
1. Your NFS share still works, so if you have log sync enabled, you should see logs under <NFS_CAM_ROOT>/wyzehacks/log, look for the first log after recent reboot (ended with "_0.log"), check the following content:
* the first couple lines shows the versions, check what's the "WyzeApp" version and "WyzeHack" version
* after the line "WyzeHack: Notifying iCamera about SD card insertion event...", search for "mmc Insert..." to see if you can find any.

2. You should still have telnet access to the camera. Try telnet into the camera and run the following command and paste the output here:
```
cat /proc/`pidof iCamera`/environ
ls -l /tmp/wyze_hack/run/bin/
cat /system/init/app_init_orig.sh
```

vrurg, 2021-03-25:
>     * he first couple lines shows the versions, check what's the "WyzeApp" version and "WyzeHack" version

Ok, this is where I should've started from the beginning. It was v0.5.02 to start with and that I should've noticed from the beginning. Now I see what happened: I didn't know that the firmware binary is not part of the repository and that it is updated with `build.sh`. 

So, no problem here. Closing. Thank you!

closed 2021-03-25 by vrurg