HclX/WyzeHacks #104: [Feature Request] Softbricked Wyzecam V3 Fix

BiatuAutMiahn, 2021-03-27:
From `/system/init/app_init.sh`:
```
DEBUG_STATUS='/configs/.debug_flag'
```

After creating this file for testing an rtsp server from [https://github.com/jmichault/ipcam-100](https://github.com/jmichault/ipcam-100) the camera will not run wpa_supplicant.

Can we have options to run wpa_supplicant if `/configs/.debug_flag` exists?

HclX, 2021-03-31:
unfortunately I think there is no easy way for you to get out of this state. Checking that script when this debug flag file exists, the entire app_init.sh is skipped.

BiatuAutMiahn, 2021-03-31:
what about injecting a line into rcS to remove the file?

HclX, 2021-03-31:
You can modify the roots to do that. However, to get the roots onto device,
your only method is sdcard recovery, which now has signature verification
enabled...

I think your only solution is to take apart the camera, solder serial wire
so you can get a login console over aerial. Once you get there, you can
login and manually delete the file.

BiatuAutMiahn, 2021-04-01:
Can I do this without compromising the ingress protection permanently?

HclX, 2021-04-01:
I'm not sure about that. I think if you do it carefully it should be fine.
There are youtube videos on how to open the case carefully.

BiatuAutMiahn, 2021-04-01:
I just got into the console, it appears that WyzeHacks did run as the root passwd was updated, can we add logic to manually run wpa_supplicant from WyzeHacks init if `/configs/.debug_flag` exists?