HclX/WyzeHacks #105: Question: telnet only?

cburns4563, 2021-03-31:
If I wanted to use "Remote install using remote_install.sh” to enable telnet and nothing else, would I simply copy config.inc.TEMPLATE to config.inc and then run remote_install.sh ? 

Thanks!

HclX, 2021-04-01:
Yes, it should work, although I never tried it.

cburns4563, 2021-04-01:
Doesn’t seem enable telnet. I can hear it clicking, no voices, and then it comes back online. I can ping it but no telnet.

No log files or anything unusual on the SD card. ( I tried it with and without the SD card. )

Thanks.

INFO:root:Serving firmware file './firmware.bin' as 'http://192.168.0.105:11808/firmware.bin', md5=7a76fd71beccdc140e232ee38fa09e81
INFO:root:Checking device, mac=2CAA8E91B1CE
192.168.0.108 - - [31/Mar/2021 20:48:37] "HEAD /firmware.bin HTTP/1.1" 200 -
192.168.0.108 - - [31/Mar/2021 20:48:37] "GET /firmware.bin HTTP/1.1" 200 -

HclX, 2021-04-01:
Are you doing this on v3 camera? For v3 if you are using master branch you
need to do v3_init.sh first

cburns4563, 2021-04-01:
No, it’s a V2.

The first thing remote_install.sh does is :

	cp ./FIRMWARE_660R.bin ./firmware.bin

I don’t appear to have FIRMWARE_660R.bin and I don’t see it anywhere in the WyzeHacks release. Am I supposed to grab this from somewhere else?

Thanks. I’ll continue poking around.

cburns4563, 2021-04-01:
Ok, got it ( telnet )  to work on a V2 camera. What I was doing wrong was I poking around in the ‘installer’ directory and not in the ‘releases' directory.

Once I discovered the releases directory, I unzipped the most recent one and ran remote_install.sh and it worked perfectly. 

I may try a V3 later today. 

Thanks for your help.

HclX, 2021-04-01:
Yep, you should use release directory. I agree it can be a little bit
confusing. I can rename the directory I future changes.

endertable, 2021-04-09:
> Ok, got it ( telnet ) to work on a V2 camera. What I was doing wrong was I poking around in the ‘installer’ directory and not in the ‘releases' directory. Once I discovered the releases directory, I [...]

Hi, You still looking for an alpha tester for your program. I’m currently not working or anything so let me know if you have not found a tester  yet I would like to volunteer.  :)

cburns4563, 2021-04-09:
> … <x-msg://442/#>

Hi,

Yes. I kinda got side tracked on making the direct live streaming work through NATs which involves something called udp hole punching. That’s just starting to work. 

Are you more interested in the cli or the live streaming?

endertable, 2021-04-09:
> 
> Hi, Yes. I kinda got side tracked on making the direct live streaming work through NATs which involves something called udp hole punching. That’s just starting to work. Are you more interested in the cli or the live streaming?
> […](#)

Hi. both are interesting but the local live streaming is something I have been very interested in with these cams (using stock firmware like you have).

PS.  How's your leg injury progressing?

cburns4563, 2021-04-09:
> … <x-msg://448/#>

Do you have V2’s or V3’s. The V3’s are a little touchy. 

For now, installation requires telnet access.

You can get here 

wget www.sonic.net/~crb/stream_hack.tgz

There’s a ReadMe file in there. And my email address is in there too. Let’s move to that.

I still can’t walk but I can drive now so that’s an improvement. Hopefully start putting ( more ) weight on it next week.

Let me know how it goes.

Thanks.

endertable, 2021-04-09:
Got the file, will start on this tonight. 😀

mrlt8, 2021-04-11:
Hey @cburns4563 just tried your stream_hack on a v2 with a fresh WyzeHacks install.

I was able to use the stream in homebridge and view the camera on an appletv.

The live stream for the camera is still functioning in the Wyze/beta app, but I'm not getting any motion notifications from Wyze for the camera with the stream_hack installed.

Do you have a repo for your stream_hack?

cburns4563, 2021-04-11:
That’s interesting about the motion notifications. Is it still capturing the event video but just not giving you a notification about it?

No repo yet, it’s still rather experimental and I wanted a one or two people to shake it down before I make a release. 

The version you have writes a rather verbose log file, that may be an issue if it runs for a long time. Check in /tmp, maybe it’s filling up.

Thanks!

mrlt8, 2021-04-11:
No events are getting recorded to the Wyze cloud on the camera with stream_hack enabled. 

Commenting out `export LD_LIBRARY_PATH=/configs/stream_hack:$LD_LIBRARY_PATH` seems to re-enable the events for the camera.

Another thing I noticed is that motion tagging (green box around motion) doesn't seem to work when stream_hack is enabled via `LD_LIBRARY_PATH`.

I also noticed that `export THIS_BIN=$(readlink -f $0)` was missing on the camera with stream_hack installed, but adding it back didn't seem to make a difference.

I'm on the beta app/firmware (4.9.6.241) if that matters and my other cams on the same firmware with and without the WyzeHacks enabled seem unaffected.

FiveLeavesLeft, 2021-04-18:
Hello @mrlt8,
The events not getting uploaded was dependent of which firmware was being run. I think I've fixed the issue and I've created a repo at https://github.com/FiveLeavesLeft/WyzeCameraLiveStream.

HclX, 2021-04-25:
I'm going to close this issue, for off topic discussion, please open new issues.

closed 2021-04-25 by HclX

nijhawank, 2021-04-28:
@endertable @ghost the link http://www.sonic.net/~crb/stream_hack.tgz results in a 404 error, do you have this file?

FiveLeavesLeft, 2021-04-28:
Look over in https://github.com/FiveLeavesLeft/WyzeCameraLiveStream. Also read the issues over there raised by endertable. I’m getting close to resolving them.

nijhawank, 2021-04-28:
I don’t see any wz command line binary line the one on YouTube video.

nijhawank, 2021-04-28:
@mrlt8 while I could see the RTSP stream on macOS with VLC, my hoobs based home bridge installation is not working. Can you provide the homebridge-ffmpeg configuration that you used to stream the camera to HomeKit.

> Hey @cburns4563 just tried your stream_hack on a v2 with a fresh WyzeHacks install.
>
> I was able to use the stream in homebridge and view the camera on an appletv.
>
> The live stream for the camera is [...]

FiveLeavesLeft, 2021-04-28:
The repo is  https://github.com/FiveLeavesLeft/WyzeCameraLiveStream <https://github.com/FiveLeavesLeft/WyzeCameraLiveStream> we should move this conversation over there.

V2 firmware versions 4.9.6.241 and 4.9.6.218 do not report events, I should have a fix for this early next week. 4.9.6.199 does not have this problem.

Thanks,

Charlie