HclX/WyzeHacks #107: v3 cameras periodically go offline

hyukishi, 2021-04-06:
I've noticed that I typically have 2 out of the 4 v3 cameras I own, all of which have WyzeHacks installed, will go offline for extended periods of time and most times have to be rebooted manually by unplugging/plugging the power.

I've also tried pinging the cams from the computer with all requests timing out. I am using the ping keepalive option with the default of 5. I lowered it to 1 for a short time. I quickly noticed the degraded performance of my 2.4ghz wireless network connection and set the ping keepalive back to 5. The problem still persists.

Suggestions? Insight? Help me Obi-Wan Kenobi!

endertable, 2021-04-08:
Are you running any custom software on them?
Are they still writing to the NFS drive even when you can't ping them (or the SD card)?

Troubleshoot by disabling the NFS write option and reboot them.  See if they write the recordings to the SD card during the whole time (even after you can't ping them anymore).  If they do, then it may be a network issue.  If you have alot of wifi devices connected to just one router, the router may not be able to handle all the devices.  Many lower and mid end routers have a device capacity of 5-20 wifi devices and so some devices may not reliably stay connected to the network or the router may not be able to handle all the network congestion.

hyukishi, 2021-04-08:
I’m running stock firmware with WyzeHacks installed over top. I haven’t implemented any custom scripts. When the cameras lose connectivity, I can still see them connected on the network devices page of my router. When I look at the NFS share, I can see just from the folder listing under the dated folder under recordings, there are entire hours here and there that did not get recorded. I’m not using SD cards with my cameras because I rely on WyzeHacks to record the data to the NFS share. They do reconnect normally after some time, but it sometimes takes more than an hour for that to happen.

Here’s a look at my directory for one of the cameras under record/20210408/

4.0K drwxr-xr-x 2 nobody 4.0K Apr 8 01:00 00/
4.0K drwxr-xr-x 2 nobody 4.0K Apr 8 02:00 01/
4.0K drwxr-xr-x 2 nobody 4.0K Apr 8 03:00 02/
4.0K drwxr-xr-x 2 nobody 4.0K Apr 8 04:00 03/
4.0K drwxr-xr-x 2 nobody 4.0K Apr 8 05:50 04/
4.0K drwxr-xr-x 2 nobody 4.0K Apr 8 06:00 05/
4.0K drwxr-xr-x 2 nobody 4.0K Apr 8 08:10 06/
4.0K drwxr-xr-x 2 nobody 4.0K Apr 8 10:12 08/
4.0K drwxr-xr-x 2 nobody 4.0K Apr 8 10:59 10/
4.0K drwxr-xr-x 2 nobody 4.0K Apr 8 11:38 11/

endertable, 2021-04-08:
When they lose connectivity, are you able to ping them from another device?  Can you telnet to them?  If not, then I  think it is still a network issue.  Some routers keep a cache of “connected” devices and those cams may not be actively connected at the time you pull up the page on the router.?.  Also, how many total devices do you have bound to that router, including all the cams, computers, phones, Tablets,  and other WiFi smart devices?  Just asking for troubleshooting purposes

hyukishi, 2021-04-08:
There’s about 40 devices connected to the router. None of them drop off of the network at all, except for the cameras. As stated in the initial posting, I cannot ping or telnet into the cameras when they’ve dropped from the network. I have tried just to be certain. The router is the ASUS RT-ACRH13. I’ve had this router for about 2 years and never had a problem with it. The only problem I have is with the cameras. And before you ask if they’re getting a strong wireless signal, the answer is yes. They have a signal strength 2-3 bars out of 3 bars.

hyukishi, 2021-04-08:
Here’s some new info:

I’ve just noticed that each time I power cycle a camera that was offline, the camera that was cycled comes back online, but it seems to kick a camera offline. They all have different MAC addresses and each one has a reserved IP on my router as well.

hyukishi, 2021-04-10:
It has to be network congestion. Turning off recording on the cameras has alleviated the issue. I believe the problem stems from the fact that my nfs share is on a wirelessly connected server due to a lack of Ethernet ports.

closed 2021-04-10 by hyukishi