HclX/WyzeHacks #108: No files in the record folder

bufordt13, 2021-04-07:
I have things set up, but I think there's something wrong with my NFS config. I get folders and files created, but the record folder gets created with 644 permission, and nothing is written inside it. If I manually change it to 744, then I get folders inside, but those folders also have 644 permission. If I keep chmod'ing the folders, it eventually starts writing video files, until it goes to the next hour, at which time the new hour folder gets 644 permissions and the Wyze cam can't write to it any more.

Here is my nfs export file:

/mnt/security_video *(rw,all_squash,insecure,async,no_subtree_check,anonuid=1003,anongid=1003)

1003 is the UID and GID of the wyzecam user I have created.

endertable, 2021-04-08:
Just for a test, can you get rid of the "anonuid=1003,anongid=1003", restart NFS and see what happens as root user?

bufordt13, 2021-04-08:
If I change it to no_root_squash, and remove the anonuid/anongid then it works. But I'm not sure how comfortable I am doing that. It seems weird that all the directories except the Record ones get the proper permissions. Maybe those are getting created by a different process on the Wyzecam.

endertable, 2021-04-08:
Your Linux may be running NFS V4 and with that it may require that that user ID matches the client user ID in order for permissions to work correctly. On the cam we only have user ID 0 which is root.  I would say if your /mnt/security_video is isolated it may not be a huge security risk.  It would be more of a security issue if you allowed passwordless SSH to a server from the cam.   I don’t run the NFS deal anymore simply because of network congestion but when I did I really didn’t have any issue with just allowing it to be root as my Linux runs NFS V4.  Good luck in whatever you decide.

bufordt13, 2021-04-08:
You're probably right on the security front. I have the cams on their own WLAN, and NFS limited to the IP range they are on.

How much more network congestion do you think there is using Wyze to NFS vs letting the WyzeCam push everything to the cloud?

endertable, 2021-04-08:
When you say “push everything to the cloud” do you mean using Wyze’s cloud service and alerts or do you mean using another cloud service?  You actually already have what I was going to do If I continue the NFS deal, which was put the cams on their own WLAN so that they didn’t interfere with all my other network activities.  I had four cams and understand that they are all set to the exact same time so every minute, on the minute, all the cams were writing their 10MB mp4 files to my NFS server at exactly the same time.  This would interfere with Xbox, video streaming, etc. because I only had one WLAN for everything.  It was more a matter of bad logistics on my part.  My solution was going to be to create more WLANs just for smart devices.   You already figured that out, so you are probably good to go.

I chose to go with a much more complicated solution so that I don’t store so much unnecessary video nor clog my existing network  😬

closed 2021-04-08 by bufordt13