HclX/WyzeHacks #109: SD card required to install on a v3 with firmware v4.36.0.280

nijhawank, 2021-04-13:
Hi, I have few v3s that are already wall mounted and updated to the version 4.36.0.280. None of them have any SD cards installed.
I wanted to understand do I need a SD card to install WyzeHacks? The config.inc.TEMPLATE suggests to put it in the SD card but the other documentation suggests that SD card based installation doesn’t work on v3s or v2s with recent firmware versions.

Can you please clarify the installation steps and whether SD cards are required?

HclX, 2021-04-16:
Nop, SD card is no longer needed. All you need is to put a config.inc file side by side to the remote_install.sh. However, there are some caveats:
1. if installation fails, having SD card will provide some useful logs for debugging purpose.
2. v3 camera needs to be in a certain state to install this hack: basically, one partition (rootfs) needs to be a specific version. There is no easy way to tell what version it currently is and the only way to ensure this version is to do an SD card recovery to some known firmware versions, such as 4.36.0.218, or later. So if your installation fails this is probably the reason, again, without logs in SD card, there is no way to know the exact failure reason.

sveip, 2021-04-16:
2. Can you check /proc for some unique info to determine footfs, like `tune2fs
-l $(grep " / " /proc/mounts | grep -v rootfs | cut -f 1 -d " ") | grep
created` e.g.?

HclX, 2021-04-16:
I can and I'm doing it right now. And that's how  wyzehack purposely failing the installation when version doesn't match. It's just there is no way to get this information out to the user...

sveip, 2021-04-17:
I meant to check the version and choose the correct hack build with the
appropriate rootfs (given it's a limited #) ;-)

nijhawank, 2021-04-20:
I checked all firmwares that are publicly available for Wyze cam v3 and all of them have the same rootfs version which is 4.36.0.112 (the same one that is modified and included in the WyzeHacks. It seems that originally shipped v3s have rootfs 4.36.0.56 which is not in any of the publicly available firmware packages.

HclX, 2021-04-20:
Yep, that's exactly how I decided to use rootfs 112 as a base image.

nijhawank, 2021-04-20:
Hi @HclX can you release a version of WyzeHacks that simply flashes the rootfs 4.36.0.112 even if the existing one is 4.36.0.56. The reason is because all of my cameras are already mounted high up there and it would be a hassle to go through SD card recovery for them (I'd really need to climb up the ladder). 

I guess as we already know that WyzeHacks works fine with modified 4.36.0.112 and it anyways has to reflash the modified one, can you introduce a flag to skip the rootfs version match validation?

HclX, 2021-04-20:
Yep, I think I'm going to do that. Just give me a couple days.

HclX, 2021-04-21:
@nijhawank added this support in bugfixes branch, please checkout the release 0.5.06 from that branch. I don't have a v3 device with that version so didn't do any full E2E test. Please let me know how it works.

nijhawank, 2021-04-21:
It worked flawlessly. Perfect. Thank you for your awesome work.

closed 2021-04-25 by HclX