HclX/WyzeHacks #110: "REBOOT_AT" using incorrect timezone

twig123, 2021-04-15:
The config.inc file states that "REBOOT_AT" is in Local Time.

I'm in the Eastern timezone (-4 UTC, and have my config set to this:

export REBOOT_AT='04:00'

I expect it to reboot at 4am Eastern, but its actually restarting at 8am Eastern.
This would mean this setting is still trying to use UTC time, Instead of Local time.

HclX, 2021-04-25:
Can you verify your camera has correct timzezone settings? For example, make sure the timestamp on the video clip matches the local time? I verified this feature works correctly on my v3 camera.

Another way to verify it is to telnet into the camera and run "date" to see what is the output.

twig123, 2021-04-26:
Apparently my camera has 2 different times...
Viewing the time via the timestamp on the cam, it is correct.
Running 'date' from telnet, the time is off by 4 hours.

I've tried the "sync time" feature of the app, but that hasn't helped.
How can the timestamp and the 'date' have 2 different times?

Any idea on how to fix 'date' so that it pulls the correct time?

HclX, 2021-04-26:
I have no idea. What version and model is your camera? Can you run "cat
/etc/TZ" and see what it shows?

twig123, 2021-04-26:
This is on a v2 cam.

Running that command, shows that TZ is UTC+8:00 (which is wrong, it should only be +4)

twig123, 2021-04-26:
Don't know if is the proper way, but I ran "vi /etc/TZ" and changed it from UTC+8, to UTC+4
Rebooted the cam, and now "date" shows the proper time. I'm going to assume this should solve the issue at hand.

twig123, 2021-04-26:
I've no idea why  "/etc/TZ" was set wrong, nor how it got set that way... (nor why the video timestamp was correct, but "date" wasn't). However, I've confirmed that manually editing it to UTC+4:00 (instead of UTC+8:00), has corrected the problem of it rebooting at the incorrect time.

closed 2021-04-26 by twig123