HclX/WyzeHacks #111: Anyone got Wyze's home monitoring thing?

HclX, 2021-04-15:
Let me know and i need some help on getting the firmware dump of the hub.

AK5nowman, 2021-04-21:
I’ve got my hands on the new kit.

HclX, 2021-04-21:
Nice, I accidentally corrupted the flash of my hub so I'd like to get a
dump of a working one to recover mine. If you are ok, can you run some
tools for me to get a dump? Everything you will be running is open source
so no need to worry about me being a bad guy 🙂

To do this you need to know how to telnet and how to run Linux commands.
Also, you will need a NFS server to store the dump, but I assume that's not
a problem if you are using wyzehack.

Let me know if you want help and I will send you instructions.

AK5nowman, 2021-04-21:
Yeah if you can send some instruction I can give it a go.

HclX, 2021-04-22:
it's good you have serial wire soldered, you still need to crack their root password to log into the console.  Anyway, here is my instructions:
1. Make sure you add your hub into your wyze account.
2. Verify the hub is running firmware version 4.32.2.120 or above
3. Clone my WyzeUpdater repo
4. On a linux machine, from the WyzeUpdater repo, run 
```
./wyze_updater.py --token ~/.wyze_token update -d <mac_addr_of_your_hub> -f firmwares/hms_telnet.bin -p 18080
```
it will ask your wyze credential, just like the wyze hack installer does, so don't worry, your credential will not go anywhere.

5. If succeeded, it will start a telnet demon. The only way to tell if it's working or not is to try telnet into your hub using username "root" and password "ismart12" (at this moment, you can probably use the same credential to log into the serial console, but i didn't try that).
6. This telnet daemon will be there for a couple minutes before the hub reboots, once you logged in, try the following command (you need to get your nfs server information ready):
```
mount -o nolock, rw <your nfs_server>:/<nfs_share> /mnt
dd if=/dev/mtdblock0 /mnt/mtdblock0 bs=4096
dd if=/dev/mtdblock1 /mnt/mtdblock1 bs=4096
dd if=/dev/mtdblock2 /mnt/mtdblock2 bs=4096
dd if=/dev/mtdblock3 /mnt/mtdblock3 bs=4096
```
7. If everything works, you get all the firmware dumps in your nfs share. These dumps don't contain any device specific information (those are in mtdblock6 and mtdblock7).  You can either upload the dump here or ping me for private emails if you like.

AK5nowman, 2021-04-22:
[WyzeHMSDump.zip](https://github.com/HclX/WyzeHacks/files/6356292/WyzeHMSDump.zip)

Here you go

HclX, 2021-04-22:
Great, I will see if I can recover mine with this dump. Btw, you now have
telnet access, even though it's temporary, you can inspect firmwares
without worrying about it the flash layout etc.

AK5nowman, 2021-04-22:
When you say it’s temporary, how temporary? Does it completely remove the installed hack or just reboot periodically? I took a quick peek in some of the squashfs and found the apps that they run. Going to toss them into ghidra today and see what’s good. 

A question I keep wondering about is, I wonder if it would be possible to pull the firmware from the cc1310 and flash it onto the v1 dongle, if that would work and let us receive data for the keypad. Looks like the keypad pairs up with the v1 dongle but doesn’t spit out anything when you click buttons on the pad. My thought was lack of support at the cc1310 level.

HclX, 2021-04-22:
I think the problem is the interface: v1 bridge is using a ch651 (or
similar) to convert the cc1310 output to hidraw packages, there might be
some restrictions on how that device works with cc1310.

HclX, 2021-04-23:
My hub is recovered, thanks for the dump!

AK5nowman, 2021-04-23:
Thats great! What are you goals with tinkering with the hub?

Seems like the cc1310 communicates over uart.  Do you think the ch554 (or w/e) actually modifies the raw data packet that is being sent? Looks like on the HMS the bridge app starts the dongle_app which opens ttys0. So far most stuff looks similar to the firmware I got from a v2 with the exception that there are  ~7 additional packet types that look specific to the keypad. I'm going to see if it's possible to take the cc1310.bin file from the hms and update my v1 dongle, without any additional tools.

HclX, 2021-04-23:
well, I guess my current goal is to make it work, and gain permanent root access.

Regarding your problem: yes, you should be able to push the cc1310 firmware to the bridge using the camera firmware. I think i saw some code doing that. But i'm not sure if if the new firmware will work with ch554. It might be worth reading the ch554 user manual and see how it converts input to USB output. Of course, if you can hook a logic analyzer onto the pins between ch554 and cc1310, it will be much easier to tell how they communicate.

closed 2021-04-25 by HclX