HclX/WyzeHacks #115: Issue with v3 camera

ydjzx, 2021-05-08:
Not exactly an issue of WyzeHacks but camera itself.  I somehow bricked one camera once and recovered by SD card flashing.  After that I cannot install WyzeHacks on to it.  The camera works fine.  Once I run remote_install.sh the purple (red+blue) light blinks and the camera goes offline.  So nothing can be pushed to it.  It stays in this status for about 15 minutes then reboots.  No install log written to sd card.  
I tried runing v3_init.sh but cannot telnet it after reboot.  I tried different versions of firmware and fatory reset but no luck.  Not sure what else I can do.

tjbcatch22, 2021-05-08:
I'm wondering what version of firmware you recovered with?  Have you tried an older version Wyze firmware and then try the hack process again?  Wyze official firmware is here: `https://wyzelabs.zendesk.com/hc/en-us/articles/360024852172`

tjbcatch22, 2021-05-08:
The oldest firmware on the wyzelabs site above looks to be 4.36.0.125.  I'm wondering what happens if you tried going back to that one?  Just an idea.

ydjzx, 2021-05-08:
I tried it as well but no luck.  Thanks for the suggestion though.
I think HclX the camera can be telneted to with some cable.  I'm just not sure what cable to use.

HclX, 2021-05-10:
@ydjzx With latest release from master branch, you don't need v3_init.sh anymore. It will automatically "initialize" the v3 camera if it hasn't been done.

ydjzx, 2021-05-10:
@gszakacs I'm considering about that but I guess I'll lose warranty by doing this.
@HclX I know that.  I'm just trying everything out of desperation.  I can't get telnet enbaled on that camera not matter how.

ydjzx, 2021-05-11:
I've tried both sd card recovery and factory reset more than once.  Didn't work.
The camera is still within return window.  I guess I'll return it and buy a new one.

HclX, 2021-05-11:
Does the camera still function?

ydjzx, 2021-05-11:
Yes works perfectly fine.

gszakacs, 2021-05-13:
@ydjzx after a bit of trial and error, I found that the issue in my case was network related.  For all other purposes my cameras / pcs worked well, but when I tried running the script to push the firmware hack, the process failed. Pretty sure somehow this was related to my vlan config (even though my linux pc and v3 cam were on the same subnet)... Anyway, after I moved the v3 camera to a separate wifi network (as basic config as it gets) and pushed the firmware update using the new network, all worked well

ydjzx, 2021-05-13:
Unfortunately I don't have extra router to setup another subnet.  Cannot test.

closed 2021-06-07 by ydjzx