HclX/WyzeHacks #119: [Feature Request] Reboot on ping Fail?

BiatuAutMiahn, 2021-05-17:
Can you implement a feature in the config that if an address cannot be ping'd (ie; 192.168.1.1) for like 60 seconds that the camera will reboot?

Scenario: I have a camera in my car as a dash cam that is connected to an AutoPi. The concern is that when the AutoPi goes to sleep due to low battery or engine is off for too long the hotspot is no longer active and I loose the NFS mount. I'd rather have the camera reboot constantly until the car is running again.

Thank you.

HclX, 2021-05-18:
Don't we already have NFS disconnect detection?

BiatuAutMiahn, 2021-05-18:
> Don't we already have NFS disconnect detection?

It does recover as far as I can tell, nonetheless this would still be a useful feature

endertable, 2021-05-21:
You can use the custom script feature Wyzehacks has and just create your own script for this.  If you don’t know shelol scripting let me know, I can lead you to some that have already been written for similar purpose.

BiatuAutMiahn, 2021-05-22:
The custom script applies only if NFS mount is active right?

endertable, 2021-05-22:
If you edit /configs/WyzeHacks.sh, youcan put a call to your script in there and end it with an ampersand to let it run in background as a daemon and monitor the network, and restart network or reboot cam as needed.  Of course, first thing in your script should be a sleep 60 to allow iCamera to do the initial network startup.

BiatuAutMiahn, 2021-05-24:
ok, thank you

closed 2021-05-24 by BiatuAutMiahn