HclX/WyzeHacks #120: add RTSP to stock fw

senposage, 2021-05-20:
is it possible to add in RTSP support while keeping the wyze functionality,maby use the sdcard as a swap buffer

endertable, 2021-05-21:
V2 cams have an official RTSP firmware from Wyze that keep the wyze functionality.  Nothing yet for V3.

senposage, 2021-05-21:
whats involved in porting it in ? RTSP is a pretty basic protocol

endertable, 2021-05-21:
I agree, I don’t know why Wyze does not offer RTSP in every single firmware they release (Actually, I do know why).  As far as anyone else being able to just add in an RTSP server to Wyze firmware, Wyze makes it very difficult because their binary creates the video devices and does not share them.

mrlt8, 2021-05-21:
There is a fork of WyzeHacks that enables streaming direct from the camera:
https://github.com/FiveLeavesLeft/WyzeCameraLiveStream

It’s not RTSP, but the stream works with VLC, homebridge, home assistant, blue iris, etc