HclX/WyzeHacks #122: Installed Telnet access, but no wyzehacks?

tehgod, 2021-05-31:
First install it seems the config file was missing, so it seems it finished installing telnet but failed after not locating the config file. Corrected config filename, re-ran the setup,  but it is not making either the wyzehacks or params directories on the wyzecam. No log file is being created, even after formatting sd card from computer (not recognized anymore in app). 

Attempted to also sd card firmware flash the camera, but same issue is persisting afterwards.

endertable, 2021-06-05:
I am not really understanding what you are after, does /configs/wyze_hacks.sh exist?  Is there anything in /media?  Are you wanting NFS or SD Card?

ikrowni, 2021-06-06:
> I am not really understanding what you are after, does /configs/wyze_hack.sh exist? Is there anything in /media? Are you wanting NFS or SD Card?

I'm currently trying to uninstall wyzehacks and when trying to run the 2 commands 

```
  cp /system/init/app_init_orig.sh /system/init/app_init.sh 
  rm /params/wyze_hack.*
```

I get an error saying

`cp: can't stat '/system/init/app_init_orig.sh': No such file or directory         [root@WyzeCam-D9C1:init]#   rm /params/wyze_hack.*                                rm: can't remove '/params/wyze_hack.*': No such file or directory`

I do see the directory for configs/wyze_hacks.sh

And I do see system/init/app_init.sh but not the app_init_orig.sh

I'm just trying to get the camera back to original running condition so I can give the damn thing away to a friend. The camera seems to be working fine but the SD card is no longer seen at all like OP.

endertable, 2021-06-06:
1.  If this is a V3 cam, then there will not be a /system/init/app_init_orig.sh, you can simply edit the /configs/wyze_hack.sh file to only have the following 4 lines:

#!/bin/sh
killall -9 telnetd
busybox telnetd &
/system/init/app_init.sh &

2.  Reboot cam and let sit a several minutes to see if it writes to SD card (if it doesn't, read below for troubleshooting SD card).  
3.  After all is working well use root/WYom2020 for login.  
4.  Once you see it is booting and writing to SD, you can  'rm -f /configs/wyze_hack.*'  to disable telnet and the hack permanently .

##################################################################################
If this is a V2 cam, then don't do any of the  above and instead do the following 2 commands and list the results back here so I can give different way.  (Be careful on what you delete or overwrite as you can brick the cam):

ls -l /system/init
cat /system/bin/app.ver
##################################################################################

**Troubleshoot SD card**
As far as the SD card not being written to, do a google search for Guiformat_X86_X64.zip and use that utility to format your SD card, then try in cam again..  Make sure you did step #1 first to disable wyze hacks.

Good luck and let me know how it turns out..

ikrowni, 2021-06-06:
1 last quick question

how do i edit the files. im using putty from windows and im trying to use nano but it says 

nano not found

endertable, 2021-06-06:
Awesome.  Just to be able to help others in the future, can I ask was just diabling Wyzehacks that reallowed writing to SD card or did you have to use Guiformat on SD card?  Thanks.

ikrowni, 2021-06-06:
Yep right after writing those 4 lines and restarting I immediately saw the SD card available again. Thanks again!

HclX, 2021-07-11:
I assume the issue is solved, closing this issue.

closed 2021-07-11 by HclX