HclX/WyzeHacks #125: Any help for overriding the root password? (Instructions are geared to someone who already doesn't need instructions

delovelady, 2021-06-07:
Quoting the README.md for 05_07

`The default root password for WyzeCam is wellknown. Since this hack will enable telnetd it may pose a new threat. To avoid that, customize the password to something only you know. You will need to uncomment a variable in the configuration file to use this feature.
`
Quoting the default config file:
`
# This changes the default root password to whatever you specified. To
# generate your own password hash, use the following command from a
# linux environment:
#   openssl passwd -1 -salt <YOUR SALT> <YOUR PASSWORD>
# Use the result to replace sample hash ($1$MYSALT$3Sy1OLRk4kTa7P6fvzwp71)
# export PASSWD_SHADOW='root:$1$MYSALT$3Sy1OLRk4kTa7P6fvzwp71:10933:0:99999:7:::'
`

Is there any chance this "example" may be replaced with a minor explanation (or supplemented in the README) for those who don't already know how to do this?

Help for "openssl passwd" is equally difficult to find.  "man openssl" shows that the option passwd is valid; that's about it; no details on what the -1 -salt options do. making the FORMAT of the PASSWD_SHADOW variable better known, would be a good start.

For the example, it would be helpful if there was a description of what goes between the colons. (What IS $1?  What IS $MYSALT?  Is that the same as YOUR SALT above? (probably.)  What are the extra colons at the end???)

The example also takes an additional turn to make this this even more cryptic.  There is a shift from YOUR SALT to MYSALT in the example, enhancing the confusion.

I think, given that the general advice in the README seems to suggest the need to do this, it should be a little less cryptic to understand.  Maybe that's just me.

HclX, 2021-06-13:
Right above the password shadow in the config template file you can find the instructions.

delovelady, 2021-06-15:
> Right above the password shadow in the config template file you can find the instructions.

My purpose here was to supplement the instruction that's not stated.  What it all means.  How it works.  The example in config.txt (though it doesn't say so) merely makes the security exposure (a little) worse by ensuring there is only one password to get into all cameras (that use that example option).  The intent here was to educate and help people make their own passwords, which is far beyond what the "instruction" provided in config.txt gives.

veverkap, 2021-06-30:
@delovelady thank you for this - it helped me a lot to share this.

HclX, 2021-07-11:
Thank you for this, feel free to add a section in README.md file and create a pull request.

delovelady, 2021-07-16:
You're welcome @HclX .  I don't know the first thing about how to update/create github projects.  I'm an old school developer (heavy emphasis on the "old" part, going back to the 60s).  But I can learn. Is there a guide to do what you say?