HclX/WyzeHacks #126: Is there a way to have a .bashrc created for the root user?

delovelady, 2021-06-07:
Perhaps this question is not specifically for this topic, but I don't know a better place to ask.  And anyway, perhaps someone here knows.  I will delete this on request if it doesn't belong.

I tried to create a .bashrc for the root user upon accessing the camera by telnet.  The result was an error stating that the directory is read-only mounted.

There are certain things I put into .bashrc (or .profile) for every system I access.  For example, "set -o vi" and disabling of color (my distaste for linux's universally unreadable color selections cannot be overstated!  Black on blue!?!?!  Really?!?!?!).  And so on.  Is there a way to accomplish this?

Thanks,
Dennis

endertable, 2021-06-08:
The cam was really not meant for user space nor user environment and so they did not build bash or any other fancy shell into their busybox binary, only sh I believe.  There are ways to kludge what you may be wanting depending on are you looking to execute things on system level (maybe /etc/profile) or user only.

HclX, 2021-07-11:
nop, it's not a full fledged linux system, bashrc is not something supported.

closed 2021-07-11 by HclX