HclX/WyzeHacks #127: Doesn't seem to work with Wyze Cam V3, 4.36.1.4

delovelady, 2021-06-07:
Not sure if this is supposed to work with v3 cameras; I couldn't find anything on it.

When I tried to put WyzeHacks 05_07 onto that 4.36.1.4 camera, it stopped communicating (and I got broken pipe and other complaints from the script).  

Light on front of camera was alternating red / blue / red / blue ..

Until I repowered it, it would not connect to Wyze (or it wasn't recognized?).  Repowering put it back to normal.

`
Device type:      Camera (WYZE_CAKP2JFUS)
Device name:      **************
Firmware version: 4.36.1.4
IP Address:       192.168.1.236

Pushing firmware to this device? [y/N]:y
INFO:root:Serving firmware file './firmware.bin' as 'http://192.168.1.5:11808/firmware.bin', md5=52fca213d3333b75e819a732ce340f2f
192.168.1.236 - - [07/Jun/2021 13:31:53] "GET /firmware.bin HTTP/1.1" 200 -
----------------------------------------
Exception happened during processing of request from ('192.168.1.236', 48723)
Traceback (most recent call last):
  File "/usr/lib/python3.8/socketserver.py", line 316, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python3.8/socketserver.py", line 347, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python3.8/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python3.8/http/server.py", line 647, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/lib/python3.8/socketserver.py", line 720, in __init__
    self.handle()
  File "/usr/lib/python3.8/http/server.py", line 427, in handle
    self.handle_one_request()
  File "/usr/lib/python3.8/http/server.py", line 415, in handle_one_request
    method()
  File "./wyze_updater.py", line 188, in do_GET
    self.wfile.write(firmware_data)
  File "/usr/lib/python3.8/socketserver.py", line 799, in write
    self._sock.sendall(b)
BrokenPipeError: [Errno 32] Broken pipe
----------------------------------------
Traceback (most recent call last):
  File "./wyze_updater.py", line 362, in <module>
    args.action(creds, args)
  File "./wyze_updater.py", line 260, in update_devices
    push_update(creds, dev_info['product_model'], mac, url, md5)
  File "./wyze_updater.py", line 163, in push_update
    return run_action(creds, model, "upgrade", mac, {"url": update_url, "md5": md5, "model": model})
  File "./wyze_updater.py", line 157, in run_action
    return device_api(
  File "./wyze_updater.py", line 140, in device_api
    raise RuntimeError('Request failed, error %s:%s' % (rsp['code'], rsp['msg']))
RuntimeError: Request failed, error 3019:DeviceIsOffline
192.168.1.236 - - [07/Jun/2021 13:31:56] "GET /firmware.bin HTTP/1.1" 200 -
----------------------------------------
Exception happened during processing of request from ('192.168.1.236', 48724)
Traceback (most recent call last):
  File "/usr/lib/python3.8/socketserver.py", line 316, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python3.8/socketserver.py", line 347, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python3.8/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python3.8/http/server.py", line 647, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/lib/python3.8/socketserver.py", line 720, in __init__
    self.handle()
  File "/usr/lib/python3.8/http/server.py", line 427, in handle
    self.handle_one_request()
  File "/usr/lib/python3.8/http/server.py", line 415, in handle_one_request
    method()
  File "./wyze_updater.py", line 188, in do_GET
    self.wfile.write(firmware_data)
  File "/usr/lib/python3.8/socketserver.py", line 799, in write
    self._sock.sendall(b)
BrokenPipeError: [Errno 32] Broken pipe
----------------------------------------
192.168.1.236 - - [07/Jun/2021 13:31:59] "GET /firmware.bin HTTP/1.1" 200 -
----------------------------------------
`

terryhonn, 2021-06-11:
I've successfully installed this to one v3 with 4.36.1.4 firmware, but can't get it to install on 2 other v3s. Same errors @delovelady 
 is reporting. Happy to flash/install/test as needed to get a fix. Thanks!

pygmymarmoset, 2021-06-24:
@delovelady I did a fresh install of 4.36.1.4 from the sd card following https://support.wyze.com/hc/en-us/articles/360031490871-How-to-flash-your-Wyze-Cam-firmware-manually and it would install after that.  Before I had gotten the same broken pipe errors you had.  The camera updated to the latest (4.36.2.5) after that and is still working.

rani-pinchuk, 2021-06-27:
Ah - I now see that PR #130 actually deals with my problem. Indeed I get the following when running `df`:
```
root@WyzeCam-EA75:~]# df 
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root                 3712      3712         0 100% /
tmpfs                    45736       652     45084   1% /dev
tmpfs                    45736       912     44824   2% /tmp
tmpfs                    45736         8     45728   0% /run
media                    45736         0     45736   0% /media
/dev/mtdblock3            3584      3584         0 100% /system
/dev/mtdblock6             384       120       264  31% /configs
tmpfs                    45736       912     44824   2% /etc
df: /mnt: Value too large for defined data type
```

HclX, 2021-07-11:
I saw similar issues on my test camera. I think this is not an issue in wyzehack but some other random network reliability issues. At this moment there is not much I can do so I'm closing this issue.

closed 2021-07-11 by HclX