HclX/WyzeHacks #128: cpu usage for iCamera process in top

nijhawank, 2021-06-15:
Not an issue but I am seeing some slowness in the telnet session and the iCamera seems to be using around 27-28% of cpu sometime even jumping to 40%. All this is while the camera is powered off (soft power off from the Wyze app).
I just wanted to check with others if this is normal cpu usage of iCamera or this is a problem with my cam.

endertable, 2021-06-15:
That’s pretty normal for V2 cam but a bit on the higher side for V3.  I think the Wyze app version of “off” is not really off, that way you can turn it back “on”.  
Are those the same numbers when cam is “on”?

nijhawank, 2021-06-15:
Sorry forgot to mention that it’s a v3. So you do think it’s on the higher side.
Yes, the cpu usage is the same what the camera turned on or off.

endertable, 2021-06-16:
That iCamera binary is a beast as it pretty much has many internal functions for the cam.  It’s Wyze’s Swiss Army knife.  The V3 cam came with a bit more CPU power but it also came with Icamera needing to do some more functions, especially with its twilight VSP.  You will also see a surge when it has to copy the video files over to SD card (or NFS drive, which can take longer).  I get that slow down, too, every so often, even locking up the terminal for a few seconds.  Wyze put just enough resources in this cam to get the necessary things done and not much more. Having a user environment was certainly not their intention, I would only really worry if you start seeing your videos not being saved.   :)

HclX, 2021-07-11:
The high cpu utilization is normal, there is not much i can do for this one. I'm closing this issue.

closed 2021-07-11 by HclX