HclX/WyzeHacks #129: Can a missing apostrophe permanently brick the camera?

delovelady, 2021-06-18:
I had a working configuration for seven cameras in my collection.  But I want to change NFS servers, so I created a config.new that included these two lines:

`#export NFS_ROOT='192.168.1.10:/mnt/SSDs/Share'
export NFS_ROOT='192.168.1.199:/wyzecams
`
The old configuration was on 192.168.1.10, and the new was to be on 192.168.1.199.  Note the missing apostrophe (stupid, stupid!!!!)

The camera was working fine.  It read the configuration file (I know, because the config.new has been removed and config.inc is updated)... but upon reboot the (V3) camera started with a red dot in the front of the screen, and no amount of resetting will restore its normal operation.

I have tried the Factory Reset approach (SD card with official firmware, power off, power on)... but the device simply will not power on.  Then, because I sometimes learn slowly, I did this to a V2 camera (still not noticing the mistake in the configuration.  (I know.  I know.  My toes are sore from kicking myself.)  The V2 camera now has a green (yellow?) light on startup, and it ever changes to blue.  I tried the (different) factory reset on that camera... also to no avail.  So it looks like I've ruined two cameras... and the damage appears to be permanent.

But I have to ask:  Will such an error brick a camera... and is there any hope in sight?  Believe me, I've tried over and over to factor reset, to no avail.

endertable, 2021-06-18:
I can say that most shells that come across an unterminated  export or variable assignment will stop dead in their tracks.  Mind you, if the script contains another apostrophe somewhere then NFS_ROOT will equal all those lines of code from apostrophe to apostrophe and as you can already tell, that will mean bad results.
just a quick question, you stated:

‘ I have tried the Factory Reset approach (SD card with official firmware, power off, power on)’

Just for clarification, while powering on you held the setup button until light turns blue?  Just clarifying the details.

What I’m not sure of, is on the V3 firmware factory restore, if it writes over WyzeHack’s modification to /etc/init.d/rcS.  If it does, there shouldn’t really be a problem, if it doesn’t then there may be a problem because rcS Will continue to call /configs/wyze_hack.sh which is not good.

As far as V2, that makes no sense to me because that should be able to restore the app_init.sh instead of keeping it a link to wyze_hack.sh.    Hmmmmm...

delovelady, 2021-06-19:
@endertable thanks for the reply.  I agree about misplaced apostrophes.  I’ve been a shell programmer (and other languages) since the mid 80s, and have had my share of misplaced punctuation problems! One would think we’d learn after a certain amount of time.  But alas!  :)

Anyway, back to the original point.  Later today I will make a YouTube video of trying to jump start one or more of these, and share the link.  Yes, I followed the instructions exactly.  For the V2, holding down the setup button the whole startup time.  The light never turns blue on the V2 cameras.  For the V3, inserting a prepared SD card in an unpowered device, then powering on.  I’ve also tried the V2 reset instructions on the V3.  Nothing has produced any apparent change.  Believe me, I’ve done this a number of times… hoping against hope that the next one would bring it back to life.  And on more than one occasion I’ve held the setup button in until my fingers ache (about 5 minutes).  No joy.  A call to Wyze support was unfruitful, of course.

pygmymarmoset, 2021-06-24:
@delovelady just to make sure you didn't do any of the same things I did wrong, you did unzip the download, and did rename the extracted file to demo_wcv3.bin ?  Also, I have read that there are issues with large (>32G?) sd cards and firmware updates, I made an 8G partition on mine and formatted that and put the firmware on the partition and it worked fine.  Worth a shot anyway!

endertable, 2021-06-25:
I @pygmymarmoset i was thinking the same thing about the demo.bin (V2) and demo_wcv3.bin (V3) mostly because light never turning blue after reboot while holding setup makes me want to think the cams are not finding a firmware file.  Hopefully he figured it out.

delovelady, 2021-06-27:
You guys are Godsends!!!  V3 camera issues are solved now, completely!  I had actually renamed the file, vut incorrectly (using dash instead of underscore).  Thanks so much! I’ll revisit the V2 issues again later this weekend.

Woo hoo, you’ve lifted my spirits significantly!  Really happy!

HclX, 2021-07-11:
I'm assuming the issue is solved, so closing this.

closed 2021-07-11 by HclX