HclX/WyzeHacks #131: Cam v3 keeps rebooting only when NFS is available

rani-pinchuk, 2021-06-26:
Very interestingly, my Wyze Cam v3 keeps rebooting **only** when NFS is available. 

In my NFS directory I can find:
```
/data/nfs/nfs/WyzeCams/7C78B22CEA75/wyzehacks/log# ls -la
total 13
drwxr-xr-x 2 root root 9 Jun 26 14:10 .
drwxr-xr-x 3 root root 4 Jun 26 14:10 ..
-rw-r--r-- 1 root root 0 Jun 26 14:06 reboot_1
-rw-r--r-- 1 root root 0 Jun 26 14:07 reboot_2
-rw-r--r-- 1 root root 0 Jun 26 14:07 reboot_3
-rw-r--r-- 1 root root 0 Jun 26 14:08 reboot_4
-rw-r--r-- 1 root root 0 Jun 26 14:09 reboot_5
-rw-r--r-- 1 root root 0 Jun 26 14:09 reboot_6
-rw-r--r-- 1 root root 0 Jun 26 14:10 reboot_7
```
The config.inc is different from the original only by the lines:
`export NFS_ROOT='192.168.0.173:/data/nfs'`

And the line of the `export PASSWD_SHADOW`.

The strange thing is that if I shutdown my nfs service the rebooting stops! However, in that case, the camera stays "red" (the led of the camera is red).

I have tried to set the `NFS_TIMEOUT` to a higher number, but it does not have any effect on the behavior.

More interestingly, if I set `NFS_ROOT` to a wrong value, then the camera is happy in its blue state. Also then I can telnet to it, and mount manually to my NFS. 

So I tried running the mount_nfs() function manually, and I got to the last lines, but got: `sendto failed: Connection refused` when running `$WYZEHACK_DIR/bin/hackutils mmc_insert`

Could you suggest how to proceed? Any thoughts why the camera reboot again and again when NFS **is** available? 

Thanks for your help,

Rani

endertable, 2021-06-27:
Do you have an SD card in it?

rani-pinchuk, 2021-06-27:
No. I tried though to leave the card in, to check if this will have any effect - in both cases, it keep rebooting if the NFS is available. In both cases,  and this also `$WYZEHACK_DIR/bin/hackutils mmc_insert` gives `sendto failed: Connection refused`.

rani-pinchuk, 2021-06-27:
I have tried the same with another new Wyze Cam V3 - just out of the box. I allowed it to update as suggested by the app, and then installed the wyzehack as explained above and got the exact same behavior.

rani-pinchuk, 2021-06-27:
Ah - I now see that PR #130 actually deals with my problem. Indeed I get the following when running `df`:
```
root@WyzeCam-EA75:~]# df 
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root                 3712      3712         0 100% /
tmpfs                    45736       652     45084   1% /dev
tmpfs                    45736       912     44824   2% /tmp
tmpfs                    45736         8     45728   0% /run
media                    45736         0     45736   0% /media
/dev/mtdblock3            3584      3584         0 100% /system
/dev/mtdblock6             384       120       264  31% /configs
tmpfs                    45736       912     44824   2% /etc
df: /mnt: Value too large for defined data type
```

However, while the PR indeed avoids the `too large for defined data type`, it still keeps rebooting (although less often - about once per minute instead of twice per minute).

rani-pinchuk, 2021-06-27:
Installing my NFS on a smaller partition (457G), I got this to work, also without PR #130. So indeed the size of my original NFS mount (3.5T) was the problem. Thanks, though, for pygmymarmoset who wrote PR #130 as this gave me the hint that the size of the mount is the issue. 

I leave this issue open, if there is the intention to adapt the hack to large partitions.

HclX, 2021-07-11:
i merged PR130 to master, and with some minor fixes. Please checkout release 0.5.08 and see if the issue is fixed.