HclX/WyzeHacks #132: remote_install.sh RuntimeError

yqin, 2021-07-13:
Wyze Cam v2 with FW: 4.9.6.241

Pushing firmware to this device? [y/N]:y
INFO:root:Serving firmware file './firmware.bin' as 'http://192.168.0.1:11808/firmware.bin', md5=11567104604de4f4cc8f4633bc6c33f4
Traceback (most recent call last):
  File "./wyze_updater.py", line 362, in <module>
    args.action(creds, args)
  File "./wyze_updater.py", line 260, in update_devices
    push_update(creds, dev_info['product_model'], mac, url, md5)
  File "./wyze_updater.py", line 163, in push_update
    return run_action(creds, model, "upgrade", mac, {"url": update_url, "md5": md5, "model": model})
  File "./wyze_updater.py", line 160, in run_action
    custom_string="", action_params=params)
  File "./wyze_updater.py", line 140, in device_api
    raise RuntimeError('Request failed, error %s:%s' % (rsp['code'], rsp['msg']))
RuntimeError: Request failed, error 3005:UnauthorizedOperation

ghagiel, 2021-07-13:
Same here on a V2 and V3, most current firmware for both (4.9.6.241 and 4.36.2.5)

MomosX, 2021-07-13:
This happened to me a few times also.  Rebooted camera and then it worked.

andyjohnschneider, 2021-07-13:
I'm also encountering this on v2,v3, and Pan.  Rebooting didn't help me.

echatman, 2021-07-14:
I'm encountering the same error on a wyze cam v3 with firmware version 4.36.1.4, and rebooting the camera didn't help. The failed request appears to be a POST to https://api.wyzecam.com/app/v2/auto/run_action

ksokhan, 2021-07-14:
Also just ran into this. Tried on V3 cam with firmware version `4.36.0.280`, which worked on a second camera just a few days ago.

LukeShootsThings, 2021-07-14:
I'm seeing the same issue.  I have one v3 on firmware version 4.36.2.5, plugin version 1.7.0.33 activation date 5/11/2021 and it's currently working with wyzehacks.  I just tried to run this on two cameras I got today, same firmware version and plugin version, and I get the 3005:UnauthorizedOperation message.

mandusm, 2021-07-16:
I'm having the same issue on WyzeCam2
Device type:      Camera (WYZE_CAKP2JFUS)
Firmware: 4.36.2.5

```
Traceback (most recent call last):
  File "/home/mandusm/Builder/WyzeHacks/installer/./wyze_updater.py", line 362, in <module>
    args.action(creds, args)
  File "/home/mandusm/Builder/WyzeHacks/installer/./wyze_updater.py", line 260, in update_devices
    push_update(creds, dev_info['product_model'], mac, url, md5)
  File "/home/mandusm/Builder/WyzeHacks/installer/./wyze_updater.py", line 163, in push_update
    return run_action(creds, model, "upgrade", mac, {"url": update_url, "md5": md5, "model": model})
  File "/home/mandusm/Builder/WyzeHacks/installer/./wyze_updater.py", line 157, in run_action
    return device_api(
  File "/home/mandusm/Builder/WyzeHacks/installer/./wyze_updater.py", line 140, in device_api
    raise RuntimeError('Request failed, error %s:%s' % (rsp['code'], rsp['msg']))
RuntimeError: Request failed, error 3005:UnauthorizedOperation
```

mandusm, 2021-07-16:
As an EXTREMELY hacky way to make this work I Changed a few lines in the `wyze_updater.py` file. 

Line 256 Specifically
```
        if not server:
            if not args.addr:
                args.addr = get_host_ip(dev_info['ip'])
            #url = build_url(args.addr, args.ssl, args.port)
            url = "http://s3-us-west-2.amazonaws.com/wuv2/upgrade/WLPP1/firmware/1.2.0.80a.bin"
            server = start_http_server(firmware_data, args.addr, args.port, args.ssl)
            logging.info("Serving firmware file '%s' as '%s', md5=%s" % (args.firmware, url, md5))
```

I won't be covering any details on how to do this, but I was able to give the Wyze camera a custom response to the dns lookup for `s3-us-west-2.amazonaws.com` to simply point to the IP address of the host the script is running on.  I used `dnsmasq` to achieve this on my network. There are many other ways to do this.

gtxaspec, 2021-07-17:
confirm, @mandusm's method does work.

virmaior, 2021-07-17:
@gtxaspec  I'm not getting it work with a WyzeV3.

1. running dnsmasq  with the router set to run all traffic through that DNS and remap s3-us-west-2.amazonaws.com to my computer
2.  url set to http://s3-us-west-2.amazonaws.com/wuv2/upgrade/WLPP1/firmware/1.2.0.80a.bin 

does the url need to be something else for WyzeV3?

nadigo, 2021-07-19:
tried with a local DNS server on mac (NEMO) but the fw update gets stuck endlessly. 
any advise ?

virmaior, 2021-07-19:
Since you have the line: 192.168.4.169 - - [18/Jul/2021 22:18:13] "GET /wuv2/upgrade/WLPP1/firmware/1.2.0.80a.bin HTTP/1.1" 200  it looks like it's getting served correctly to your device by your computer.

1. does your camera reboot?
2. did you try using telnet to login to the camera? (if it's a v3 it won't make any noises indicating success).

nadigo, 2021-07-19:
thanks - it did work! I can telnet to the camera.

virmaior, 2021-07-19:
yes, but does telnet let you enter a username / password?  If so, you've already succeeded in adding wyzehacks. the default password for V3 is WYom2021 . Did you try that?

nadigo, 2021-07-19:
I get user/pass prompt but can't login with root/WYom2021. 

`Trying 192.168.4.177...
Connected to 192.168.4.177.
Escape character is '^]'.

WyzeCamV3-7F10 login: root
Password: 
Login incorrect`

virmaior, 2021-07-19:
username: root
password: YWom2021

nadigo, 2021-07-19:
can't login yet I have the telnet prompt .. 
I used fw FIRMWARE_660R.bin

ksokhan, 2021-07-19:
Unless the password changed very recently, I believe pass should be WYom2020 for v3

HclX, 2021-07-20:
I think I saw a beta version v3 firmware with different root password. Wyze is definitely trying to block this (which I totally understand). I can get the hash so if someone has a powerful gpu can run hash cat to get the new root password.

endertable, 2021-07-21:
Team, there is a big Wyze update coming out next week that may try to stop all these hacks.  They are trying to force even the app to be updated so tread lightly before you update your cams to latest firmware as it might close up all the insecurities that these hacks work on, including the update URL method that WyzeHacks uses.  I’m still on version 228 for V3 but they may force cam firmware updates, too, which they systematically can.  :(

julxb, 2021-07-21:
Hello folks,
Does this hack work on v2 with Firmware version: 4.9.6.241?
I had couple of error and fixed it with helps above, but now I don't have any error anymore but seems that it's stucked in a loop.

[image]

Am I missing something? What can I check?

* Cam : V2
* Firmware :  4.9.6.241
* Python version :  python3 -V
Python 3.7.3
* Using a Raspberry Pi 2
* NFS : mounted (192.168.X.X:/nfsshare  7.1G  1.7G  5.1G  25% /mnt/nfs - Is there a way to check if it works btw?)

Let me know,
Thanks,

virmaior, 2021-07-21:
can't use remote_install.sh to hack this because it will set the port 11808 and then that won't match what you're asking the camera to update against (port 80)

instead run the wyze_updater directly : 
sudo python3 ./wyze_updater.py --token ~/.wyze_token --debug update
-m WYZEC1-JZ -m WYZECP1_JEF -m WYZE_CAKP2JFUS -m WYZEDB3 -f ./firmware.bin -p 80

julxb, 2021-07-21:
Thanks a lot for help.
I get the 200 ok answer but seems to still stuck :(
[image]

C1ARKGABLE, 2021-07-21:
> I think I saw a beta version v3 firmware with different root password. Wyze is definitely trying to block this (which I totally understand). I can get the hash so if someone has a powerful gpu can run hash cat to get the new root password.

@HclX Lemme take a crack at it if you've got it.

virmaior, 2021-07-22:
@julxb  neither of those lines is an indication that the camera is trying to download a firmware from your computer.  It should be a GET request with the path to the firmware that you sent in the modified version of wyze_updater.py .

Are you sure you've DNS spoofed your wyze cam and not just your computer?
For starters you can see if you've spoofed the DNS on your own computer by trying a traceroute there on s3-us-west-2.amazonaws.com . it should return the IP of the computer you're trying to update the firmware from.
No way to directly test spoofing from an unhacked camera but you will need to have the DNS server set at the router level to accomplish this.

HclX, 2021-07-22:
Today I received the official push notification of the v2 update, and of course they have changed their root password. Here are the ones I'm seeing:
v2 (4.9.7.798): root:$6$wyzecamv2$hvp6M6S2JI7vyyHfEDPWCYJ8N2r5B4ZzS8uZYwyMkoWyg90sCJzupBGD57CObtKonld0Yvr2B/ejt4l4/jryi.:10933:0:99999:7:::

v3 (4.36.3.19 beta): root:$6$wyzecamv3$8gyTEsAkm1d7wh12Eup5MMcxQwuA1n1FsRtQLUW8dZGo1b1pGRJgtSieTI02VPeFP9f4DodbIt2ePOLzwP0WI0:0:0:99999:7:::

@C1ARKGABLE can you help on those passwords? Look at the salt they are taking that seriously so these passwords might not be easy to crack.

Thanks

C1ARKGABLE, 2021-07-22:
[image]

@HclX see ya in 2053... Oof

Anyone have a good dictionary to use? Or maybe a quantum computer? I'm asking for a friend...

jdkadel, 2021-07-22:
I'm getting the 3005:UnauthorizedOperation error as well when trying to remote install v0_5_08 on a v2 Cam Pan.

[This link](https://github.com/nblavoie/wyzecam-api/issues/9) has info about the API password is that what is needed?

md5(md5(md5(password)))

virmaior, 2021-07-22:
@jdkadel  NO, that has nothing to do with this.
You need to 
1. modify wyze_updater.py to replace the url line with url = "http://s3-us-west-2.amazonaws.com/wuv2/upgrade/WLPP1/firmware/1.2.0.80a.bin"
2. Spoof the DNS so that your wyze cam thinks your local computer is  s3-us-west-2.amazonaws.com (change the DNS server on your local network to accomplish this )
3. Run wyze_updater.py directly to use port 80 (on my osx system I had to use sudo at the front but here's the basic command): python3 ./wyze_updater.py --token ~/.wyze_token --debug update
-m WYZEC1-JZ -m WYZECP1_JEF -m WYZE_CAKP2JFUS -m WYZEDB3 -f ./firmware.bin -p 80  

If you run it successfully, you'll see [IP of camera] - - [18/Jul/2021 22:18:13] "GET /wuv2/upgrade/WLPP1/firmware/1.2.0.80a.bin HTTP/1.1" 200

jdkadel, 2021-07-23:
Thanks, just trying to help.  I modified the .py file and verified that the DNS changes with a ping to ...amazonaws and it returned the local computer address.  I didn't get a line with the "GET ...200".  Here's the end of the command.  I didn't get the response that the request was received.  The camera did not reboot or was it accessible on SSH.  I'll run thru the process again Sunday.

header: Connection: keep-alive
DEBUG:urllib3.connectionpool:https://api.wyzecam.com:443 "POST /app/v2/auto/run_action HTTP/1.1" 200 182
DEBUG:root:{'ts': 1627012580075, 'code': '1', 'msg': '', 'data': {'session_id': '7e3d336635934cd293b884317d687cda', 'custom_string': '', 'action_session_id': 'f1be254581204f4b87eb00ffca8a8e36', 'result': 2}}
Press Ctrl+C when all the updates are done...

virmaior, 2021-07-23:
1. it won't ever work with ssh because there's no sshd on the device. Instead, you have to use telnet or netcat. on osx, this looks like: nc 192.168.3.101 23  (substitute ip as appropriate)
2. Can you confirm that the camera is also operating from your spoofed DNS server and not just a change to the DNS on your computer (e.g. editing /etc/hosts on a desktop or laptop will only change the DNS for that local computer)?
3. The camera should reboot either when it successfully gets the wyze_updater delivered firmware _or_ after a few minutes automatically. If neither is happening...

nadigo, 2021-07-23:
> yes, but does telnet let you enter a username / password? If so, you've already succeeded in adding wyzehacks. the default password for V3 is WYom2020 . Did you try that?
> 
> **Note that I miswrote this initially as WYom2021**

WYom20200 worked. Thank you!
Making sure I understand, to enable stream I just need to add  "export STREAM_HACK=1" to /configs/wyze_hack.cfg and reboot?

jdkadel, 2021-07-26:
> 1. it won't ever work with ssh because there's no sshd on the device. Instead, you have to use telnet or netcat. on osx, this looks like: nc 192.168.3.101 23  (substitute ip as appropriate)
> 2. Can [...]

Thanks @virmaior!, I verified everything again and now it's working.   I didn't have the dnsmasq working correctly.  Thanks!!

miptton, 2021-07-30:
Thanks to everyone who contributed to the workaround instructions.
I was able to use Pihole in a container on my server to spoof the appropriate DNS addresses and install the hack.

mpatton125, 2021-07-30:
Thanks to everyone who contributed to the workaround instructions.
I was able to use Pihole in a container on my server to spoof the appropriate DNS addresses and install the hack.

47bob47, 2021-07-31:
Thanks for getting this working but can we have some baby steps?
Are my assumptions correct for a **v3 camera**:
 1.  Change line 256 in code to url = "http://s3-us-west-2.amazonaws.com/wuv2/upgrade/WLPP1/firmware/1.2.0.80a.bin"
 2.  Install a local dns server so that s3-us-west-2.amazonaws.com points to the computer from which you will be running wyze_updater.py
 3.  Set up a webserver on the computer in item 2 above and add /upgrade/WLPP1/firmware/ to the webroot?
 4.  Put firmware.bin or maybe FIRMWARE660R.bin into the webroot and alias it to 1.2.0.80a.bin  ?
 5.  After making sure your config.inc file has your necessary changes run: sudo python3 ./wyze_updater.py --token ~/.wyze_token --debug update -m WYZEC1-JZ -m WYZECP1_JEF -m WYZE_CAKP2JFUS -m WYZEDB3 -f ./firmware.bin -p 80

Obviously a little confused about that 1.2.0.80a.bin file since the file seems to be accessed via ./firmware.bin in the wyze_updated.py options string in item 4.
If local dns is going to point s3-us-west-2.amazonaws.com to ny local computer anyway, can't I just put in my local computer's address there instead of the amazon adress?

Feel free to flame me if I have everything wrong

virmaior, 2021-07-31:
1. wyze_updater.py runs its own web server and does the aliasing so you don't need your 3 or 4.
2. the aliasing is because it has to match the url of firmware that exists when the camera checks with wyze
3.  "If local dns is going to point s3-us-west-2.amazonaws.com to ny local computer anyway, can't I just put in my local computer's address there instead of the amazon adress?" **NO**, because wyzeapi now checks before sending the "update firmware" command to the camera. It has a list of known good firmwares and apis.

47bob47, 2021-07-31:
Thanks.  I guess I'm one step closer.
Set up the dns server to properly point the amazonaws domain to my local computer - all other domains work normally which I checked from both the computer that I am running the upgrade script from as well as my cell phone (turned off all data except wifi).

Getting the error message "RuntimeError: Request failed, error 3019:DeviceIsOffline" even though I am viewing my V3 wyze cam at the same time that I am running the script.

**Full error message here:**
*****
Device type:      Camera (WYZE_CAKP2JFUS)
Device name:      elevator
Firmware version: 4.36.0.228
IP Address:       192.168.1.127

Pushing firmware to this device? [y/N]:y
INFO:root:Serving firmware file './firmware.bin' as 'http://s3-us-west-2.amazonaws.com/wuv2/upgrade/WLPP1/firmware/1.2.0.80a.bin', md5=d2c262c195b31fa6f0d1fa0491bb413b
Traceback (most recent call last):
  File "./wyze_updater_CHANGED.py", line 363, in <module>
    args.action(creds, args)
  File "./wyze_updater_CHANGED.py", line 261, in update_devices
    push_update(creds, dev_info['product_model'], mac, url, md5)
  File "./wyze_updater_CHANGED.py", line 163, in push_update
    return run_action(creds, model, "upgrade", mac, {"url": update_url, "md5": md5, "model": model})
  File "./wyze_updater_CHANGED.py", line 157, in run_action
    return device_api(
  File "./wyze_updater_CHANGED.py", line 140, in device_api
    raise RuntimeError('Request failed, error %s:%s' % (rsp['code'], rsp['msg']))
**RuntimeError: Request failed, error 3019:DeviceIsOffline**
^CException ignored in: <module 'threading' from '/usr/lib/python3.8/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 1388, in _shutdown
    lock.acquire()
KeyboardInterrupt: 
******

**TEST PINGS TO AMAZONAWS AND TO GOOGLE:**
bob@LG:~/Downloads/wyze/v3/20210731_0.5.08$ **ping google.com**
PING google.com (142.250.80.46) 56(84) bytes of data.
64 bytes from lga34s34-in-f14.1e100.net (142.250.80.46): icmp_seq=1 ttl=117 time=12.7 ms
^C
--- google.com ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 12.667/12.667/12.667/0.000 ms

bob@LG:~/Downloads/wyze/v3/20210731_0.5.08$ **ping s3-us-west-2.amazonaws.com**
PING s3-us-west-2.amazonaws.com (192.168.1.71) 56(84) bytes of data.
64 bytes from 192.168.1.71 (192.168.1.71): icmp_seq=1 ttl=64 time=0.027 ms
64 bytes from 192.168.1.71 (192.168.1.71): icmp_seq=2 ttl=64 time=0.039 ms
^C
--- s3-us-west-2.amazonaws.com ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1007ms
rtt min/avg/max/mdev = 0.027/0.033/0.039/0.006 ms

**IFCONFIG TEST:**
bob@LG:~/Downloads/wyze/v3/20210731_0.5.08$ ifconfig
lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 13445  bytes 1857288 (1.8 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 13445  bytes 1857288 (1.8 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlp0s20f3: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.71  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::8acd:991e:6bc6:4b35  prefixlen 64  scopeid 0x20<link>
        ether 18:26:49:73:48:8f  txqueuelen 1000  (Ethernet)
        RX packets 998554  bytes 1151929646 (1.1 GB)
        RX errors 0  dropped 15  overruns 0  frame 0
        TX packets 342410  bytes 78755422 (78.7 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

47bob47, 2021-07-31:
UPDATE.
I didn't touch anything, but it worked this time.  I think the big difference is that previously on the phone app the camera was showing as offline but it was actually online if you clicked on it.   So camera needs to display as online on the main page of the phone app for this to work.

Just to keep this for historical purposes, here is a slightly abbreviated debug session (deleted lines that were related to other cameras):

bob@LG:~/Downloads/wyze/v3/20210731_0.5.08$ sudo python3 ./wyze_updater_CHANGED.py --token ~/.wyze_token --debug update -m WYZEC1-JZ -m WYZECP1_JEF -m WYZE_CAKP2JFUS -m WYZEDB3 -f ./firmware.bin -p 80
[sudo] password for bob: 
INFO:root:Trying saved credentials from /home/bob/.wyze_token.
DEBUG:root:{}
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): api.wyzecam.com:443
send: b'POST /app/v2/device/get_device_list HTTP/1.1\r\nHost: api.wyzecam.com\r\nUser-Agent: okhttp/3.8.1\r\nAccept-Encoding: gzip, deflate\r\nAccept: */*\r\nConnection: keep-alive\r\nContent-Length: 495\r\nContent-Type: application/json\r\n\r\n'
send: b'{"access_token": "lvtx.b9+rVvyUqVpz2QvnQZDxdRwUQGYG8AI71iyeNPjLC6g3lcR+p/yVnhZr88J3RKhcr9ZQg6/xIPjOwDelb+Q+TH22+xsVF2ohV6/KcCwn52VNP3VKEXomGUNdsMckkywwXi804FsBE7W+AeQTzuOw5rXJIweS5APKNebG8WzQh+g0kicMVDuNnS4DCmRGJ0GPSDniLA==", "app_name": "com.hualai", "app_version": "2.11.40", "phone_system_type": "2", "app_ver": "com.hualai___2.11.40", "phone_id": "04a39c87-695c-4f18-81d0-e4b84160da9c", "sc": "a626948714654991afd3c0dbd7cdb901", "sv": "f0ef3f988133430aaf14f8f00add6d2d", "ts": 1627768622000}'
reply: 'HTTP/1.1 200 OK\r\n'
header: Access-Control-Allow-Headers: Origin, X-Requested-With, Content-Type, Cache-Control, Accept, authorization, content-type
header: Access-Control-Allow-Methods: POST,GET,OPTIONS
header: Access-Control-Allow-Origin: *
header: Cache-Control: no-cache
header: Content-Type: application/json; charset=utf-8
header: Date: Sat, 31 Jul 2021 21:57:02 GMT
header: Expires: -1
header: Pragma: no-cache
header: Server: Microsoft-IIS/8.5
header: X-AspNet-Version: 4.0.30319
header: X-Powered-By: ASP.NET
header: Content-Length: 13281
header: Connection: keep-alive
DEBUG:urllib3.connectionpool:https://api.wyzecam.com:443 "POST /app/v2/device/get_device_list HTTP/1.1" 200 13281
DEBUG:root:{'ts': 1627768623241, 'code': '1', 'msg': '', 'data': {'device_list': [{'mac': '2CAA8E1BD93B', 'first_activation_ts': 1563214346000, 'first_binding_ts': 1563214348000, 'enr': 'fKv3tXln4D7Bvk5T', 'nickname': 'DV1', 'timezone_name': 'America/New_York', 'product_model': 'WYZECP1_JEF', 'product_model_logo_url': 'https://s3-us-west-2.amazonaws.com/wyze-file/device-logo/small/wyze_icon_device_pan.png', 'product_type': 'Camera', 'hardware_ver': '0.0.0.0', 'firmware_ver': '4.10.7.798', 'user_role': 1, 'binding_user_nickname': 'MY_USERNAME', 'conn_state': 1, 'conn_state_ts': 1627758065627, 'push_switch': 0, 'device_params': {'p2p_id': '7MHJKM1HSG8XE23K111A', 'p2p_type': 3, 'ssid': 'DISCO_VOLANTE', 'ip': '192.168.1.5', 'power_switch': 1, 'temperature': '0', 'humidity': '0', 'temp_humi_room_type': 1, 'comfort_standard_level': 2, 'is_temperature_humidity': '0', 'records_event_switch': 1, 'motion_alarm_switch': 1, 'audio_alarm_switch': 1, 'smoke_alarm_switch': 1, 'co_alarm_switch': 1}, 'is_in_auto': 0, 'event_master_switch': 1, 'parent_device_mac': '', 'parent_device_enr': '', 'binding_ts': 1620493820000, 'timezone_gmt_offset': -4.0}, {'mac': '7C78B21A93D5', 'first_activation_ts': 1617392801000, 'first_binding_ts': 1617392808000, 'enr': 'CFPBGXoTlzB+e7fE', 'nickname': '36_roof_level', 'timezone_name': 'America/New_York', 'product_model': 'WYZE_CAKP2JFUS', 'product_model_logo_url': 'https://wyze-file.s3-us-west-2.amazonaws.com/system-logo/device/small/wyze_icon_cam_v3.png', 'product_type': 'Camera', 'hardware_ver': '0.0.0.0', 'firmware_ver': '4.36.0.252', 'user_role': 1, 'binding_user_nickname': 'MY_USERNAME', 'conn_state': 1, 'conn_state_ts': 1627768312383, 'push_switch': 1, 'device_params': {'p2p_id': 'GUJ5EKRUBHVD9LME111A', 'p2p_type': 3, 'ssid': 'ddwrt_36_3', 'ip': '192.168.1.28', 'power_switch': 1, 'temperature': '0', 'humidity': '0', 'temp_humi_room_type': 1, 'comfort_standard_level': 2, 'is_temperature_humidity': '0', 'records_event_switch': 1, 'motion_alarm_switch': 1, 'audio_alarm_switch': 0, 'smoke_alarm_switch': 0, 'co_alarm_switch': 0}, 'is_in_auto': 0, 'event_master_switch': 1, 'parent_device_mac': '', 'parent_device_enr': '', 'binding_ts': 1617730692000, 'timezone_gmt_offset': -4.0}, {'mac': '7C78B21A9082', 'first_activation_ts': 1617481713000, 'first_binding_ts': 1617481715000, 'enr': 'ZPJd9wZZn0i7yS6+', 'nickname': 'TMC_03', 'timezone_name': 'America/New_York', 'product_model': 'WYZE_CAKP2JFUS', 'product_model_logo_url': 'https://wyze-file.s3-us-west-2.amazonaws.com/system-logo/device/small/wyze_icon_cam_v3.png', 'product_type': 'Camera', 'hardware_ver': '0.0.0.0', 'firmware_ver': '4.36.0.228', 'user_role': 1, 'binding_user_nickname': 'MY_USERNAME', 'conn_state': 0, 'conn_state_ts': 1627741210359, 'push_switch': 0, 'device_params': {'p2p_id': 'BY1VK194UDNV82TM111A', 'p2p_type': 3, 'ssid': 'Squidward', 'ip': '192.168.1.193', 'power_switch': 1, 'temperature': '0', 'humidity': '0', 'temp_humi_room_type': 1, 'comfort_standard_level': 2, 'is_temperature_humidity': '0', 'records_event_switch': 1, 'motion_alarm_switch': 1, 'audio_alarm_switch': 0, 'smoke_alarm_switch': 0, 'co_alarm_switch': 0}, 'is_in_auto': 0, 'event_master_switch': 1, 'parent_device_mac' [... cut ...]  '91', '32': '77', '33': '1', '34': '11472', '35': '3588', '36': '11229', '37': '3115', '38': '0', '39': '0', '40': '0', '41': '0', '42': '0', '43': '0', '44': '0', '45': '1', '46': '30', '47': '2', '48': '50', '49': '0', '50': '1', '51': '1', '52': '1440', '53': '151964', '54': '255', '55': '300', '56': '0', '57': '0', '58': '4578844', '59': '0', '60': '0', 'F2': '1', 'F3': '1'}, 'timezone_gmt_offset': -4.0}}

Press Ctrl+C when all the updates are done...
...^C
INFO:root:Stopping http server...

HclX, 2021-08-01:
I made a partial fix in `bugfixes` branch based on the information provided here. The only difference with my fix is it's using another wyze's update server name (`d1fk93tz4plczb.cloudfront.net`) so make sure you setup your DNS spoofing with that name. The benefit of this name is it allows ports other than 80 so you don't need to run installer script with `sudo`.

The fix *seems* to be working fine with v3 camera. However, I can't get it work with v2 camera so there might be something else wrong.

47bob47, 2021-08-01:
As I wrote before, the first camera updated successfully.
I have a second camera that I am trying to update.   As long as the wyze android app's homepage says the camera is online the lights all flash and the update completes without error using the amazonaws server.   Unfortunately the update doesn't ever stick - cannot telnet to that camera and it doesn't write to the nfs mount.  Probably tried at least 10 times.

When I try the cloudfront server I get a different error: **"RuntimeError: Request failed, error **3005:UnauthorizedOperation"****

Device type:      Camera (WYZE_CAKP2JFUS)
Device name:      37_roof
Firmware version: 4.36.0.228
IP Address:       192.168.1.193

Pushing firmware to this device? [y/N]:y
INFO:root:Serving firmware file './firmware.bin' as 'http://d1fk93tz4plczb.cloudfront.net/wuv2/upgrade/WLPP1/firmware/1.2.0.80a.bin', md5=3ffc17d39052784b6d0b6dcf447d29eb
DEBUG:root:{'provider_key': 'WYZE_CAKP2JFUS', 'action_key': 'upgrade', 'instance_id': '7C78B21A9082', 'custom_string': '', 'action_params': {'url': 'http://d1fk93tz4plczb.cloudfront.net/wuv2/upgrade/WLPP1/firmware/1.2.0.80a.bin', 'md5': '3ffc17d39052784b6d0b6dcf447d29eb', 'model': 'WYZE_CAKP2JFUS'}}
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): api.wyzecam.com:443
send: b'POST /app/v2/auto/run_action HTTP/1.1\r\nHost: api.wyzecam.com\r\nUser-Agent: okhttp/3.8.1\r\nAccept-Encoding: gzip, deflate\r\nAccept: */*\r\nConnection: keep-alive\r\nContent-Length: 784\r\nContent-Type: application/json\r\n\r\n'
send: b'{"access_token": "lvtx.b9+rVvyUqVpz2QvnQZDxdRwUQGYG8AI71iyeNPjLC6g3lcR+p/yVnhZr88J3RKhcr9ZQg6/xIPjOwDelb+Q+TH22+xsVF2ohV6/KcCwn52VNP3VKEXomGUNdsMckkywwXi804FsBE7W+AeQTzuOw5rXJIweS5APKNebG8WzQh+g0kicMVDuNnS4DCmRGJ0GPSDniLA==", "app_name": "com.hualai", "app_version": "2.11.40", "phone_system_type": "2", "app_ver": "com.hualai___2.11.40", "phone_id": "04a39c87-695c-4f18-81d0-e4b84160da9c", "sc": "a626948714654991afd3c0dbd7cdb901", "sv": "011a6b42d80a4f32b4cc24bb721c9c96", "ts": 1627822526000, "provider_key": "WYZE_CAKP2JFUS", "action_key": "upgrade", "instance_id": "7C78B21A9082", "custom_string": "", "action_params": {"url": "http://d1fk93tz4plczb.cloudfront.net/wuv2/upgrade/WLPP1/firmware/1.2.0.80a.bin", "md5": "3ffc17d39052784b6d0b6dcf447d29eb", "model": "WYZE_CAKP2JFUS"}}'
reply: 'HTTP/1.1 200 OK\r\n'
header: Access-Control-Allow-Headers: Origin, X-Requested-With, Content-Type, Cache-Control, Accept, authorization, content-type
header: Access-Control-Allow-Methods: POST,GET,OPTIONS
header: Access-Control-Allow-Origin: *
header: Cache-Control: no-cache
header: Content-Type: application/json; charset=utf-8
header: Date: Sun, 01 Aug 2021 12:55:26 GMT
header: Expires: -1
header: Pragma: no-cache
header: Server: Microsoft-IIS/8.5
header: X-AspNet-Version: 4.0.30319
header: X-Powered-By: ASP.NET
header: Content-Length: 74
header: Connection: keep-alive
DEBUG:urllib3.connectionpool:https://api.wyzecam.com:443 "POST /app/v2/auto/run_action HTTP/1.1" 200 74
DEBUG:root:{'ts': 1627822526543, 'code': '3005', 'msg': 'UnauthorizedOperation', 'data': {}}
Traceback (most recent call last):
  File "./wyze_updater_CHANGED.py", line 365, in <module>
    args.action(creds, args)
  File "./wyze_updater_CHANGED.py", line 263, in update_devices
    push_update(creds, dev_info['product_model'], mac, url, md5)
  File "./wyze_updater_CHANGED.py", line 163, in push_update
    return run_action(creds, model, "upgrade", mac, {"url": update_url, "md5": md5, "model": model})
  File "./wyze_updater_CHANGED.py", line 157, in run_action
    return device_api(
  File "./wyze_updater_CHANGED.py", line 140, in device_api
    raise RuntimeError('Request failed, error %s:%s' % (rsp['code'], rsp['msg']))
RuntimeError: Request failed, error 3005:UnauthorizedOperation

HclX, 2021-08-01:
The CloudFront one requires a different URL path. You can take a look at the latest change on bugfixes branch. Everything is built in with remote_indtall.sh already.

47bob47, 2021-08-01:
I'm starting to think there is something wrong with this camera.  Maybe that is why it shows as offline in the android app 75% of the time (even though after clicking on it, the camera is actually online). 
I tried the new versions of remote_install.py and wyze_updater.py  from bugfixes and it went through the process successfully with blinky lights and all but once again the update did not stick - no telnet and no writing to nfs.   Still functions fine as a standard wyze camera though.   I'll get another v3 camera tomorrow so I can test with something else.

After the camera is back in online status on the wyze app I will put a "--debug" into the remote_install.py script to see if it gives any other information.

Here is the output without "--debug":

Found local config file, including into firmware update archive...
Upgrade/config.inc
Requirement already satisfied: requests in /usr/lib/python3/dist-packages (2.22.0)
INFO:root:Trying saved credentials from /root/.wyze_token.
INFO:root:Checking device, mac=7C78B21A9082

Device type:      Camera (WYZE_CAKP2JFUS)
Device name:      37_roof
Firmware version: 4.36.0.228
IP Address:       192.168.1.193

Pushing firmware to this device? [y/N]:y
INFO:root:Serving firmware file './firmware.bin' as 'http://d1fk93tz4plczb.cloudfront.net:11808/UpgradeKit/firmware.bin', md5=18d17731ecbdb231bfcd0e4a1aa0d665
**192.168.1.193 - - [01/Aug/2021 12:23:04] "GET /UpgradeKit/firmware.bin HTTP/1.1" 200 -**
INFO:root:Checking device, mac=2CAA8E1BD93B

47bob47, 2021-08-01:
**With --debug enabled:**

INFO:root:Checking device, mac=7C78B21A9082
DEBUG:root:{'device_mac': '7C78B21A9082', 'device_model': 'Unknown'}
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): api.wyzecam.com:443
send: b'POST /app/v2/device/get_device_info HTTP/1.1\r\nHost: api.wyzecam.com\r\nUser-Agent: okhttp/3.8.1\r\nAccept-Encoding: gzip, deflate\r\nAccept: */*\r\nConnection: keep-alive\r\nContent-Length: 552\r\nContent-Type: application/json\r\n\r\n'
send: b'{"access_token": "lvtx.HAHNy94z1qnJU0LKsBrJbafo97fhAJQGIxRo9YOiukdRhqllDSfLxCFS0mudgTa8Kknfz35V/q4x3rlxEUc/Jr7WpGcyhsNKuwcdR9dPvuXvoJ92O/1wPCzurv4+5DdZdy82f0zeopp4LGUiqmWlS0Hb2Xwc5WYIi9HtQ7CS6N5J0cKJzAKv8IKB8YDezJcPtPRTrQ==", "app_name": "com.hualai", "app_version": "2.11.40", "phone_system_type": "2", "app_ver": "com.hualai___2.11.40", "phone_id": "291ed65c-c3c3-467f-8b59-7a9cba653c67", "sc": "a626948714654991afd3c0dbd7cdb901", "sv": "81d1abc794ba45a39fdd21233d621e84", "ts": 1627838886000, "device_mac": "7C78B21A9082", "device_model": "Unknown"}'
reply: 'HTTP/1.1 200 OK\r\n'
header: Access-Control-Allow-Headers: Origin, X-Requested-With, Content-Type, Cache-Control, Accept, authorization, content-type
header: Access-Control-Allow-Methods: POST,GET,OPTIONS
header: Access-Control-Allow-Origin: *
header: Cache-Control: no-cache
header: Content-Type: application/json; charset=utf-8
header: Date: Sun, 01 Aug 2021 17:28:06 GMT
header: Expires: -1
header: Pragma: no-cache
header: Server: Microsoft-IIS/8.5
header: X-AspNet-Version: 4.0.30319
header: X-Powered-By: ASP.NET
header: Content-Length: 1742
header: Connection: keep-alive
DEBUG:urllib3.connectionpool:https://api.wyzecam.com:443 "POST /app/v2/device/get_device_info HTTP/1.1" 200 1742
DEBUG:root:{'ts': 1627838887324, 'code': '1', 'msg': '', 'data': {'mac': '7C78B21A9082', 'dtls': 0, 'first_activation_ts': 1617481713000, 'first_binding_ts': 1617481715000, 'enr': 'FGOy3kRj69wMrVGI', 'nickname': '37_roof', 'timezone_name': 'America/New_York', 'product_model': 'WYZE_CAKP2JFUS', 'product_type': 'Camera', 'hardware_ver': '0.0.0.0', 'firmware_ver': '4.36.0.228', 'user_role': 1, 'binding_user_nickname': 'MY_USERNAME', 'parent_device_mac': '', 'parent_device_enr': '', 'binding_ts': 1627775316000, 'p2p_id': 'BY1VK194UDNV82TM111A', 'p2p_type': 3, 'ssid': 'Squidward', 'ip': '192.168.1.193', 'property_list': [{'pid': 'P1', 'value': '1', 'ts': 0}, {'pid': 'P1001', 'value': '1', 'ts': 1627837818000}, {'pid': 'P1002', 'value': '0', 'ts': 1627837818000}, {'pid': 'P1003', 'value': '0', 'ts': 1627837818000}, {'pid': 'P1004', 'value': '0', 'ts': 1627837818000}, {'pid': 'P1005', 'value': '1', 'ts': 1627837818000}, {'pid': 'P1006', 'value': '0', 'ts': 0}, {'pid': 'P1007', 'value': '0', 'ts': 0}, {'pid': 'P1008', 'value': '0', 'ts': 0}, {'pid': 'P1010', 'value': '1', 'ts': 0}, {'pid': 'P1014', 'value': '0', 'ts': 0}, {'pid': 'P1017', 'value': '0', 'ts': 0}, {'pid': 'P1018', 'value': '0', 'ts': 0}, {'pid': 'P1019', 'value': '0', 'ts': 1627789271235}, {'pid': 'P1020', 'value': '1', 'ts': 0}, {'pid': 'P1021', 'value': '1', 'ts': 0}, {'pid': 'P1022', 'value': '0', 'ts': 0}, {'pid': 'P1033', 'value': '0', 'ts': 1627837818000}, {'pid': 'P1047', 'value': '1', 'ts': 0}, {'pid': 'P1048', 'value': '1', 'ts': 0}, {'pid': 'P1049', 'value': '0', 'ts': 0}, {'pid': 'P1050', 'value': '0', 'ts': 0}, {'pid': 'P1056', 'value': '0', 'ts': 0}, {'pid': 'P18', 'value': '0', 'ts': 0}, {'pid': 'P1901', 'value': '0', 'ts': 0}, {'pid': 'P2', 'value': '0', 'ts': 1627775316535}, {'pid': 'P3', 'value': '1', 'ts': 1627837818000}, {'pid': 'P4', 'value': '1', 'ts': 0}, {'pid': 'P5', 'value': '1', 'ts': 1627837813264}], 'device_setting': {}, 'timezone_gmt_offset': -4.0}}

Device type:      Camera (WYZE_CAKP2JFUS)
Device name:      37_roof
Firmware version: 4.36.0.228
IP Address:       192.168.1.193

Pushing firmware to this device? [y/N]:y
INFO:root:Serving firmware file './firmware.bin' as 'http://d1fk93tz4plczb.cloudfront.net:11808/UpgradeKit/firmware.bin', md5=21556c52ddaa5ba9f3532d7861db6405
DEBUG:root:{'provider_key': 'WYZE_CAKP2JFUS', 'action_key': 'upgrade', 'instance_id': '7C78B21A9082', 'custom_string': '', 'action_params': {'url': 'http://d1fk93tz4plczb.cloudfront.net:11808/UpgradeKit/firmware.bin', 'md5': '21556c52ddaa5ba9f3532d7861db6405', 'model': 'WYZE_CAKP2JFUS', 'firmware_ver': '9.9.9.9'}}
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): api.wyzecam.com:443
send: b'POST /app/v2/auto/run_action HTTP/1.1\r\nHost: api.wyzecam.com\r\nUser-Agent: okhttp/3.8.1\r\nAccept-Encoding: gzip, deflate\r\nAccept: */*\r\nConnection: keep-alive\r\nContent-Length: 799\r\nContent-Type: application/json\r\n\r\n'
send: b'{"access_token": "lvtx.HAHNy94z1qnJU0LKsBrJbafo97fhAJQGIxRo9YOiukdRhqllDSfLxCFS0mudgTa8Kknfz35V/q4x3rlxEUc/Jr7WpGcyhsNKuwcdR9dPvuXvoJ92O/1wPCzurv4+5DdZdy82f0zeopp4LGUiqmWlS0Hb2Xwc5WYIi9HtQ7CS6N5J0cKJzAKv8IKB8YDezJcPtPRTrQ==", "app_name": "com.hualai", "app_version": "2.11.40", "phone_system_type": "2", "app_ver": "com.hualai___2.11.40", "phone_id": "291ed65c-c3c3-467f-8b59-7a9cba653c67", "sc": "a626948714654991afd3c0dbd7cdb901", "sv": "011a6b42d80a4f32b4cc24bb721c9c96", "ts": 1627838891000, "provider_key": "WYZE_CAKP2JFUS", "action_key": "upgrade", "instance_id": "7C78B21A9082", "custom_string": "", "action_params": {"url": "http://d1fk93tz4plczb.cloudfront.net:11808/UpgradeKit/firmware.bin", "md5": "21556c52ddaa5ba9f3532d7861db6405", "model": "WYZE_CAKP2JFUS", "firmware_ver": "9.9.9.9"}}'
[... cut ...]
header: Content-Length: 182
header: Connection: keep-alive
DEBUG:urllib3.connectionpool:https://api.wyzecam.com:443 "POST /app/v2/auto/run_action HTTP/1.1" 200 182
DEBUG:root:{'ts': 1627838892155, 'code': '1', 'msg': '', 'data': {'session_id': '3aa50b3e27d74359a15da58d04f0582c', 'custom_string': '', 'action_session_id': '788a2355f536442abfc32811e8420dfa', 'result': 2}}
DEBUG:root:request received, path=/UpgradeKit/firmware.bin
**192.168.1.193 - - [01/Aug/2021 13:28:13] "GET /UpgradeKit/firmware.bin HTTP/1.1" 200 -**

HclX, 2021-08-02:
@47bob47 the log shows your camera is getting the hack firmware but there might be issues installing the hack. Can you put an SD card in the camera to collect install.log?

47bob47, 2021-08-02:
No install log created on the sd card, just the "normal" wyze directories alarm/record/time_lapse and file time.ini.

47bob47, 2021-08-03:
UPDATE
I got a new v3 from the amazon yesterday.  I was able to update it after several tries.
Still cannot update the other one - same as before - it looks like it is updating but the update doesn't persist.

I have already reflashed the firmware to 4.36.0.228.   I think that for the next step I will try to reflash the firmware via the android app and then manually bring it down to 4.36.0.228.  Is 4.36.0.228 the latest version which works with this WyzeHack?

jk111, 2021-08-03:
The DNS spoofing method is not working on my v2 cam running 4.9.7.798 with 05.08 wyzehacks. Running tcpdump on my DNS server, I don't ever see a request for s3-us-west-2.amazonaws.com from the camera. I can confirm the camera is using my local DNS server as it makes tons of requests on bootup.

I do see the cam making requests for wyze-upgrade-service.wyzecam.com when I try to trigger remote_install.sh on it.

HclX, 2021-08-03:
I think there is something different on v2 when dealing with update. If
someone can sniff the update request sent by app, I can probably get it
work.

jk111, 2021-08-04:
Are you looking for a network trace or something else? I can get the network trace but looking at it yesterday it's some DNS lookups followed by lots of HTTPS traffic.

jk111, 2021-08-05:
So I was able to do some investigation with the v2 cam. Using a cam that I haven't upgraded yet and still has telnet access, I modified the cacert file to add my own CA, created and signed a cert for d1fk93tz4plczb.cloudfront.net, which prevents the cam from sending a TCP RST to my webserver.

Looking at the logs on my webserver, the camera is requesting https://d1fk93tz4plczb.cloudfront.net/UpgradeKit/1625707411/Upgrade_4.9.7.798_07071601.tar

Hope this helps some

j-justice, 2021-08-07:
I was able to set up the DNS redirect and ran the install script and it goes through each camera and then goes back to waiting for a new command. No recordings are showing up and I've tried rebooting the cameras too. I've also made sure to update the NFS location but I'm not sure how to tell it actually installed because I don't get any audible sound saying it's installing/installed. Any advice?

47bob47, 2021-08-08:
Hi j-justice - As you can probably tell from this thread the wyze hack has become a moving target with wyze continually trying to prevent new boot firmware from being installed onto the hardware they produce.
As far as I can tell, cameras made after Sept 2020 or so are much harder to hack.
I have a couple of v2 and v3 cameras sitting in front of me that just won't load the new firmware.  Great cameras that are missing a lot of functionality.
Much respect for the guys on this thread and the folks working on the github who are trying to get this to work.

virmaior, 2021-08-08:
@i-justice 
1. v3 make no noise on install
2. if you can telnet to the device (regardless of whether you can successfully login), then it did successfully install. Can you telnet them?

j-justice, 2021-08-08:
@virmaior I tried using telnet to login and got the below message...
`pi@raspberrypi:/ $ telnet 192.168.1.16
Trying 192.168.1.16...
telnet: Unable to connect to remote host: Connection refused`

I also don't get any message after any camera
https://i.imgur.com/5jqrHs9.png
https://i.imgur.com/1wbeJcw.png

jk111, 2021-08-09:
@HclX here is the POST from app.. if you need to see specific values from it, let me know. Took anything out I thought might be sensitive. MAC address is in format 2CAA00112233

{
	"access_token": "token-val",
	"app_name": "com.hualai",
	"app_ver": "com.hualai___2.23.21",
	"app_version": "2.23.21",
	"phone_id": "aaaaaaaa-aaaa-aaaa-aaaa-aaaaaaaaaaaa",
	"phone_system_type": "2",
	"sc": "32-character-value",
	"sv": "32-character-value",
	"ts": 13-character-value,
	"provider_key": "WYZEC1-JZ",
	"action_key": "upgrade",
	"instance_id": "CAM-MAC",
	"custom_string": "CAM-MAC",
	"action_params": {
		"url": "https:\/\/d1fk93tz4plczb.cloudfront.net\/UpgradeKit\/1625707411\/Upgrade_4.9.7.798_07071601.tar",
		"md5": "6bb9aceedc951c9f2a4190da8af57385",
		"model": "WYZEC1-JZ",
		"firmware_ver": "4.9.7.798"
	}
}

HclX, 2021-08-10:
> @HclX <https://github.com/HclX> here is the POST from app for v2
> upgrade.. if you need to see specific values from it, let me know. Took

I got the same traffic on my capture. I think there is something in the new
firmware preventing it from working. We will have to wait for wyze to
release a new firmware to see what the app sends...

mpatton125, 2021-08-10:
I had to rollback to a prior software version, and then the spoofing operation worked.

HclX, 2021-08-10:
Exactly. There must be something on the latest firmware side restricting
this behavior.

47bob47, 2021-08-10:
So, which version did you roll back to?

[48 more replies]