HclX/WyzeHacks #135: WyzeUpdater not work for some of my cameras

widewing, 2021-07-24:
I have 6 wyze v2 cameras and they all running same rtsp firmware. I tried to hack them by the wyzeupdater, the result is confusing: two of them successfully hacked and I'm able to login via telnet, the other 4 not work although the script have no errors, just never received the wget requests.

I do noticed the recent api change and the need of spoofing dns, and I did that already btw.

endertable, 2021-07-29:
I had a problem with Wyehacks on some of the newer V2 cams running the latest RTSP software, too.  Do they all have the same FCC ID?

widewing, 2021-07-30:
@endertable I purchased them in different time, and they have "almost same" FCC ID: 2ANJHWYZEC2 and 2ANJHWYZEC**0**2. Two of them are purchased together and the same style sticker and same FCC ID, but one can be hacked another can not.

endertable, 2021-07-31:
Then you may be having the issues that are mentioned in another issue posting.  Try these at https://github.com/HclX/WyzeHacks/issues/132#issuecomment-881757205

widewing, 2021-07-31:
@endertable actually I tried this already, and that's how I got the two of my devices work.