HclX/WyzeHacks #136: Question: Does wyzehacks keep deleting oldest recordings on low NFS space?

ashjas, 2021-08-07:
I have been seeing this behaviour.
Is that correct? I havent enabled ARCHIVE_OLDER_THAN flag in config.
I observer that slowly the oldest folders are reducing in size as the space left reaches below something 1GB.
Looks to be intentional, and not a bad thing, but its not documented. Please confirm.

HclX, 2021-08-08:
The hack doesn't do it. Deleting old recordings is a feature of wyze's firmware. Your NFS is mounted as an SD card so it behaves like an SD card.

However, if you enabled auto archive, archived recordings should not be deleted.

BiatuAutMiahn, 2021-08-10:
Can we set a quota on the hack?

HclX, 2021-08-21:
It's possible, but you can easily do that with nfs quota I guess.