HclX/WyzeHacks #138: Official Wyze V3 RTSP firmware support?

satmandu, 2021-09-23:
RTSP firmware has now been released for the Wyze V3: https://download.wyzecam.com/firmware/rtsp/demo_v3_RTSP_4.61.0.1.zip
https://www.reddit.com/r/wyzecam/comments/pu4hij/wyze_cam_v3_rtsp_firmware_beta_test_9232021/

Any chance of this getting this supported so we can do things like enable USB ethernet?

jassycliq, 2021-11-15:
Thanks to @HclX, I was able to use the tools he provided to create a version that works with the RTSP firmware.
Just a reminder that you have to be running the firmware already in order for it to work. [This is the link](https://github.com/jassycliq/WyzeHacks/releases/tag/v0.6.0)

evanheckert, 2021-11-16:
I thought they'd blocked the DNS spoof method that allows this to be installed?

jassycliq, 2021-11-16:
From my understanding:

- In firmware, Wyze started forcing https and also removed telnet
- In the API, Wyze is now checking the urls provided to the cameras for the upgrades

@HclX Has done some great work and has a working update that you have to download separately from their [WyzeUpdater repo](https://github.com/HclX/WyzeUpdater)

Finally, the following command is what I used alongside their updated WyzeUpdater:
`./wyze_updater.py \
    --token ~/.wyze_token \
    update -d <camera_mac> -f directory/to/hacked/firmware.bin \
    --url-host 's3-us-west-2.amazonaws.com' \
    --url-path 'wuv2/upgrade/WYZE_CAKP2JFUS/firmware/4.36.3.19.tar' \
    -p 18080`

evanheckert, 2021-11-16:
I haven't figured out how to successfully create the DNS spoof sadly. I'll keep googling for instructions. I'm embarrassed that I'm needing my hand held so much. 

So the order of operations at this point is:
1. install factory RTSP firmware (done and working)
2. Download WyzeUpdater master and extract
3. Download the v0.6.0 you linked to above and extract
4. Put config.inc and firmware.bin into the Upgrade folder
5. Turn on DSN spoofing
6. Run the command you shared above, swapping out the camera's mac address, and change the firmware director to '''./Upgrade/firmware.bin''' 

And that should use wyzeupdater to install the 0.6.0 firmware?

jassycliq, 2021-11-16:
I would turn on your DNS spoofing as soon as possible. I'm not too familiar with networking but I've had issues with dns caching in the past. Other than that, what you wrote looks good up to the confi.inc portion, that goes on root of SDCard.

For my DNS spoofing, I use PiHole as my networks DNS server and then just added the s3 domain to the local DNS record.

evanheckert, 2021-11-16:
@jassycliq Thanks for the replies! I couldn't see how to do it in Adguard Home, so spun up a Pihole instance and a traceroute shows that calls to aws server redirect to my laptop's IP Address. 

I run this command: 
```
./wyze_updater.py --token ~/.wyze_token update -d "7c:78:b2:94:0c:4d" -f /Users/evanheckert/Downloads/WyzeUpdater-master/Upgrade/firmware.bin --url-host 's3-us-west-2.amazonaws.com' --url-path 'wuv2/upgrade/WYZE_CAKP2JFUS/firmware/4.36.3.19.tar' -p 18080
```
I've tried with and without the doublequotes around the Mac address, but still get:
```
INFO:root:Trying saved credentials from /Users/evanheckert/.wyze_token.
INFO:root:Checking device, mac=7c:78:b2:94:0c:4d
Traceback (most recent call last):
  File "/Users/evanheckert/Downloads/WyzeUpdater-master/./wyze_updater.py", line 433, in <module>
    args.action(creds, args)
  File "/Users/evanheckert/Downloads/WyzeUpdater-master/./wyze_updater.py", line 268, in update_devices
    dev_info = get_device_info(creds, mac)
  File "/Users/evanheckert/Downloads/WyzeUpdater-master/./wyze_updater.py", line 148, in get_device_info
    return device_api(creds, URL_V2_GET_DEVICE_INFO, SV_V2_GET_DEVICE_INFO, device_mac=mac, device_model=model)
  File "/Users/evanheckert/Downloads/WyzeUpdater-master/./wyze_updater.py", line 132, in device_api
    raise WyzeApiError(rsp)
__main__.WyzeApiError: {'ts': 1637038554880, 'code': '3001', 'msg': 'DeviceInfoNotExist', 'data': {}}
```
I'm completely certain that the MAC address is correct, as both Opnsense and Ruckus Unleashed recognize a new device with manufacturer Wyze with the same IP Address that the Wyze app uses in the RTSP address. The RTSP stream works as well.

Any thoughts?

jassycliq, 2021-11-16:
Don't use the colons for your mac address and its without double quotes.

evanheckert, 2021-11-16:
egads, progress! 

The light is now solid purple, with a brief moment off every second. I haven't heard the voice say "installing" or anything, but the CLI is still adding dots. Would that suggest that the camera is not finding the served file?

jassycliq, 2021-11-16:
Hmmm, not to sure. For me it only takes a few seconds to finish.

`..........................................................................` is what my console shows by the time it finishes.

evanheckert, 2021-11-16:
Another step - so check opnsense logs, and it was making it through to amazon - apparently my AP's interface had a DNS override to AdGuard Home then google, so it wasn't hitting pihole. Fixed that, and this time the log showed up in Pi-hole as:
```
Type: AAAA
Domain: a24rq1e5m4mtei.iot.us-west-2.amazonaws.com
Client: 192.168.1.145
Status: OK (cache) 
Reply: IP (0.0ms)
```
And while it took about the same amount of dots as you just shared, I got no voice feedback as expected. Though I just tried `telnet 192.168.1.145` and got `wyze-car-cam login:`, which means it worked, right?

jassycliq, 2021-11-16:
> Another step - so check opnsense logs, and it was making it through to amazon - apparently my AP's interface had a DNS override to AdGuard Home then google, so it wasn't hitting pihole. Fixed that, [...]

Yup telnet isn't started on original fw, login info is in one of the other threads on mobile now so can't link.

evanheckert, 2021-11-29:
So now that we've had success installing on the factory RTSP firmware, is there a working method for ethernet usage?

Vendo232, 2021-12-29:
any idea what could be wrong?

[image]

Vendo232, 2021-12-29:
I was able to enable temporary Telnet using this process. I wanted to install WyzeHack Stream and it did install it but I assume due to temporary Telnet it does not work after reboot.

I`m unable to enable the telnet permanently per this instruction

To make it permanent, all you need is telnet into it and modify /system/init/app_init.sh by removing the comment before line /system/init/run_telnet.sh &

because the app_inst.sh is REDA ONLY. any idea what to do?

jassycliq, 2021-12-29:
>  @evanheckert 
> I was able to enable temporary Telnet using this process. I wanted to install WyzeHack Stream and it did install it but I assume due to temporary Telnet it does not work after reboot.
> 
> I`m unable to enable the telnet permanently per this instruction
> 
> To make it permanent, all you need is telnet into it and modify /system/init/app_init.sh by removing the comment before line /system/init/run_telnet.sh &
> 
> because the app_inst.sh is REDA ONLY. any idea what to do?
> 

Which instructions are you following?
Which firmware version are you on?
Which command did you run?
Which repo are you using?

Vendo232, 2021-12-29:
> Which instructions are you following?

I did this

install factory RTSP firmware (done and working)
Download WyzeUpdater master and extract [WyzeUpdater repo ](https://github.com/HclX/WyzeUpdater)
Download the v0.6.0 you linked to above and extract [( This is the link )](https://github.com/jassycliq/WyzeHacks/releases/tag/v0.6.0) 
Put config.inc and camera_telnet.bin into the Upgrade folder
Turn on DNS spoofing
Run this command
`
./wyze_updater.py --token /home/pi/.wyze_token. update -d D03F27XXXXXX -f /home/pi/WyzeUpdater/Upgrade/camera_telnet.bin.bin --url-host 's3-us-west-2.amazonaws.com' --url-path 'wuv2/upgrade/WYZE_CAKP2JFUS/firmware/4.36.3.19.tar' -p 18080 `

jassycliq, 2021-12-29:
> > Which instructions are you following?
> 
> I did this
> 
> install factory RTSP firmware (done and working)
> Download WyzeUpdater master and extract [WyzeUpdater repo ](https://github.com/HclX/WyzeUpdater)
> Download the v0.6.0 you linked to above and extract [( This is the link )](https://github.com/jassycliq/WyzeHacks/releases/tag/v0.6.0) 
> Put config.inc and camera_telnet.bin into the Upgrade folder
> Turn on DNS spoofing
> Run this command
> `
> ./wyze_updater.py --token /home/pi/.wyze_token. update -d D03F27XXXXXX -f /home/pi/WyzeUpdater/Upgrade/camera_telnet.bin.bin --url-host 's3-us-west-2.amazonaws.com' --url-path 'wuv2/upgrade/WYZE_CAKP2JFUS/firmware/4.36.3.19.tar' -p 18080 `
> 
> Camera went into flashing mode, display was showing flashing progress using the dots ( never ending ) and while dots were growing the Telnet became enabled but only READONLY which as I`m reading now is normal for V3.
> 
> Question: Which firmware.bin should I use with version 0.6?

That intended with the firmware you used. 
If you're trying to flash the firmware I posted, just input the correct file path. You extracted it and ran the correct commands but used the wrong firmware bin.

revenant-81, 2021-12-30:
There is probably some hardware differences between the recent V3 and the older V3s. 

I was able to install the WyzeHack onto the V3 RTSP firmware for devices where the MAC starts with "2C:AA:xx" using the WyzeUpdater python script with DNS spoofing. Telnet was available after reboot without adjustments. 

The more recent V3s purchased from HomeDepot had a MAC of "D0:3F:xx". I was not able to install WyzeHack onto these devices regardless of downgrading firmware.

evanheckert, 2022-01-08:
I got telnet working, but not clear how to make it use ethernet instead of wifi.

Vendo232, 2022-01-08:
I would like to know steps how to enable LAN as well

inthroxify, 2022-01-22:
> The more recent V3s purchased from HomeDepot had a MAC of "D0:3F:xx". I was not able to install WyzeHack onto these devices regardless of downgrading firmware.

I have D0:3F units recently purchased from HD, and I couldn't get it to install until I tried a smaller SD card. I could install the RTSP firmware first with the larger card, but after that, I needed a smaller card in the unit with the .inc file present, to apply the hacks. Maybe try a smaller SD card, if you have one, @revenant-81

revenant-81, 2022-01-24:
@inthroxify - I was using a 32GB Sandisk. I'll try to dig out a smaller one and give it a shot. Thanks!