HclX/WyzeHacks #141: TypeError: 'str' object is not callable

alwzfolwd, 2021-11-14:
im trying ./remote_install.sh on a Wyze Cam Pan running Firmware version 4.10.8.501 Plugin version 1.2.26.21. everything went fine and dandy until...

┌──(toxictech㉿Master)-[~/Downloads]
└─$ ./remote_install.sh
Found local config file, including into firmware update archive...
Upgrade/config.inc
Requirement already satisfied: requests in /usr/lib/python3/dist-packages (2.25.1)
INFO:root:Trying saved credentials from /home/toxictech/.wyze_token.
INFO:root:No saved credentials found, logging in with username/password...
Please enter the account name:*****************.com
Please enter the password:*****
Using phone SMS for 2FA
Enter the verification code:******
verification code: ******
INFO:root:Credentials saved to /home/toxictech/.wyze_token.
Traceback (most recent call last):
  File "/home/toxictech/Downloads/./wyze_updater.py", line 362, in <module>
    args.action(creds, args)
TypeError: 'str' object is not callable

Not sure if this is a script error (havent gotten that far yet) or if, as i just realized, the sdcard is unavailable. However, i replaced the SD and the problem persists. Would this happen if say the SD slot has malfunctioned, or perhaps i am way off. But, thats why I'm here. :) Much appreciated!

Toxic

alwzfolwd, 2021-11-22:
so...

virmaior, 2022-02-19:
probably an issue with a special character in your username or password not getting escaped properly.