HclX/WyzeHacks #146: Wyze v3: Initial setup without Wyze App possible?

iointerrupt, 2022-02-09:
I just unboxed my Wyze v3 cam and I want to use it without internet on my own IoT (internet-less) VLAN with RTSP streaming and without the use of the Wyze App. I'm looked at a lot of unofficial firmware/workaround repos for any guide that will allow me to initally setup the camera on my local wifi but all of them require you to use the Wyze v3 app (which means install wyze app, sign up for a wyze account, and intialize the camera with their cloud service first).

virmaior, 2022-02-19:
the wyze device sets up through the wyze app. The device reads a QR code produced by the app. I've not checked the contents of the QR code, but at a minimum it contains the SSID / password. I'd suspect it also contains the username and some sort of key.

iointerrupt, 2022-04-08:
Yeah the QR code generates a string in the following format:
b=[RANDOM ALPHANUMERIC]&s=[BASE 64 ENCODED SSID]=&p=[BASE 64 ENCODED AND RANDOM ENCODED SSID PASSWORD]&t=-6&r=USA&ty=1

The **b** parameter is random and the password is encrypted as well (possibly using the value of b as a salt??).  The funny thing is, I noticed a lot of other cheap cameras with their own app use the same format (Yi Home App, 360 Camera APP).

After the scanning and when the camera connects to the wifi, the camera still needs to communicate with the App and do some further initializations like liking with the wyze account to the cloud.

I was also able to use the same QR code 2x to connect the camera to the wifi which means the qr code is probably time based.

kohrar, 2022-09-17:
This appears to be possible if you use a side loaded hack that utilizes the `factory_t31_ZMC6tiIDQN` boot method such as wz_mini_hacks. 

The hurdles with a completely offline approach are:
1. Firmware updates / downgrades - You will have to do this manually using the SD card method. 
2. Setting up the initial WiFi connection - The initial WiFi configuration still has to be done with a previously generated QR code from the App. Once you are connected, you'll have to modify the .user_config file and set `bindOk=1` so it connects with the saved wifi password and ssid.
3. Camera configuration and settings - Some settings that can be changed in the Wyze App can be changed manually once you can SSH/telnet in. Settings like these include the green box on objects or enabling the Wyze logo, etc. These options can be set by editing `/configs/.user_config` manually.  Some of these options / settings are [documented on my personal notes here](https://leo.leung.xyz/wiki/Wyze_Cam#Camera_configuration). Reboot the camera or restart iCamera for these to take effect.

I was poking away at making the Wyze Cam v3 run without internet and [came up with a binary based patch for iCamera](https://github.com/gtxaspec/wz_mini_hacks/issues/38#issuecomment-1250000186). As part of my testing, I was able to set up a new Wyze Cam v3 (straight out of a box, never connected to the cloud) without using the Wyze App or having the Wyze Cam connect to the internet at all. 

Here is what I did to setup a Wyze Cam as a RTSP/IP camera without touching the cloud or internet (copied from my [Wiki](https://leo.leung.xyz/wiki/Wyze_Cam#Setup_a_Wyze_Cam_without_the_Wyze_App_or_internet)):
1. Unbox the Wyze Cam v3
2. Place the latest firmware file on a SD card (I used 4.36.9.139) and insert it to the Wyze Cam
3. Hold the SETUP button and power on the device. Hold the button down until you see both blue and red lights on
4. After the upgrade is done, the camera will reboot and begin flashing the red LED. This red LED probably means the device isn't set up and is awaiting you to do something. Power off the camera.
5. Insert the SD card containing wz_mini_hacks into the camera. When you set up the SD card, ensure that your SSH keys are set in wz_mini/etc/ssh/authorized_keys. We will need to SSH in once the camera comes up to finish up the setup.
6. Power on the camera. You should hear the wz mini hacks initializing message.
7. Press the SETUP button once. You should hear it stating it is ready for a QR code.
8. Present it with a previously generated QR code with your WiFi settings. 
9. The camera should join your wireless network and you should be able to SSH into it at this point.
10. SSH in and ensure that:
* Your WiFi settings are saved in /configs/.wifissid and /configs/.wifipasswd
* Edit the /configs/.user_config file and ensure the bindOK=0 line is set to bindOK=1. This will allow the camera to use these config files on boot. Without this, the camera will reboot back into the blinking red state.
* If you want RTSP, edit /opt/wz_mini/wz_mini.conf and ensure RTSP is enabled.
11. Reboot the camera and verify it can come up and connect to your network.