HclX/WyzeHacks #151: Remote install : Error 3005:UnauthorizedOperation

Sokonomi, 2022-12-29:
I have followed the instructions but Remote install failed.

Camera is Cam v2 running 4.9.5.115 firmware
Host is Linux Mint from virtualmachine.

virmaior, 2022-12-29:
WyzeHacks isn't going to work with that firmware version.

Unfortunately, they've blocked this particular route, because that version validates the SSL of the firmware url. For a brief window of time, it would accept a port 80 OTA firmware update which allowed for DNS spoofing to serve arbitrary firmwares.

Try https://github.com/gtxaspec/wz_mini_hacks
which uses a different method (special filename on SD card) to load a hack.

Sokonomi, 2022-12-29:
I was just reading up on the DNS spoof method, which is unfortunate and somewhat difficult to deal with..

I've tried the wyzehack SD card flash method as well, which seems to have opened telnet atleast, but its not generating any NFS mounts.

I shall take a look at the wz mini hack, thank you.

Sokonomi, 2022-12-29:
Went to take a look at wz mini hack, but that just lets you mount through a script, and I have no idea how any of that is supposed to work. I need a more casual hack, like Wyzehack (was).