HclX/WyzeHacks #158: Doorbell camera (white with bell icon over led ver 1?)

bm16ton, 2023-12-25:
Hello,
        found a wyze doorbell cam on clearence at home depot. It refused to read qr code. So i soldered to uart headers. U-boot was unlocked so i moded the bootargs bypassing init and going to shell. Finished the init by hand and added my ssid and password to user_config and wpa_conf. It connects to wifi booting straight to shell or normal boot. Didnt know about any of the hacks till after. The apps partition is rw and persists. My quick driveby of getting telnetd working resulted in it accepting my connection but not bringing up the login or shell. Unfortunatly it seems the root password is different then v2 or v3 of cams. I have jack working on the shadow but no dictionarys so im not optimistic about time or even coming thru. Strangly telnetd doesnt start in regular boot, i threw it into the rc script in /system/init also in the wpa script, nada. Also nmap sees nothing when in regular boot. I dont have a lot of time invested in it yet. Id like to be able to access the camera thru some sorta standard api or ioctl but boot logs threw the word mipi around and all i could see using the camera was a binary which i doubt the source is available for. Like most vendor implimementations im sure the kernel source (i assume they a have gpl package with kernel sources) is packed with closed source drivers with any luck they have the binarys and .o's so i can recompile my own and with a lil patching (depending what closed source) sniff the calls to the camera. Interestingly  they have a few usb kernel drivers built in pl2303, ch340, and for reasons id love to know zydas 1211 usb 802.11b/g wifi card (of all wifi cards why?) but didnt see anything bout mass storage. So my next steps in no particular order are get something like strace copied over along with ssh and maybe sum rtsp stuff ripped from other devices firmware, see what comes out of the usb to uart devices and also throw the zydas on because i havent used it in over a decade. And checkout the kernel source situation maybe get mass storage going along with other stuff. It definitely has a usb hub attached, i think it has two usb ports so otg could possibly work, if only one port sol. I can post the steps to boot via uart commands and get online (a little lengthy so wasnt sure if appropriate), but its not provisioned because i havent setup a wyze account nor any clue where that login info would be saved. I see the hardware is most like ver3 of cam except the apps is jffs2 and rw so id hafta modify for that if wyzehacks. Also doesnt look like anything from camera is stored on device only streamed which is fine for me. They do have binarys for network storage included. They also have a hostapd conf file but i didnt see hostapd anywhere. Just took a quick look at the gpl release and iw/iwtools is listed but no iw/iwconfig stuff is anywhere i can see. Maybe sum stuff is still hidden in initrd, i cant login with normal boot to see until i get the root password. Did you guys get ver 2 ver 3 passwords by brute or some other means? Ill keep on it, any questions or instructions dont hesistate to ask. Awesome project btw, very impressive!

virmaior, 2023-12-25:
woah boy.

Good work. 

The older PWs got leaked or hacked somehow for V2 and V3. 

There's a better easier way to hack these. Basically, the bootloader will automatically boot up `factory_t31_ZMC6tiIDQN` from an SD card.

Hop on over to https://github.com/gtxaspec/wz_mini_hacks

and see what we've go so far.

bm16ton, 2023-12-25:
unfortunately the doorcams, or at least mine lacks aa sdcard, all we have is a usb port that comes hidden under a sticker on the back. We also have aa cc1310 sub ghz radio to control the wireless chime. I bbelieve those are the big differences between the hardware, tho i have no idea if they reused same camera or not. Looks like same realtek wifi. Strangely the boot logs show it finding and initializing an mmc but i assume sum internal storage just presents that way, but all i see are mtd parts from the spi flash. I havent looked into it. A hidden file will cause firmware upgrade and the script simply unmounts the related partition to be upgraded, wipes it clean then blindly copies everything from another partition (where the upgraded firmware is resting) over then remounts. So seems easy enuff to install new parts. For time being i figure ill keep stock and just add sum things, get a handle on it all. Without sdcard no reason for wyze to release a downloadable firmware for it. Ill make disk images of all the parts and post them on github sometime soon in  case anyone needs them for recovery/experimenting.

bm16ton, 2023-12-25:
Ok so kernel source link for doorcam downloads the cam ver 2 kernel source. I Downloaded the cam ver3 kernel source used its config and compiled the required scsi and usb mass storage modules to get usb storage working. Here they are along with list of commands that init the system when booting init=/bin/sh https://github.com/bm16ton/wyze-doorcam-dump .

rdaigle007, 2023-12-25:
Amazing judo skills and tactics.   

Wish I had such skills to hack into my new Meta Smartglasses.   Haven’t heard of anybody hacking into these…. I had dreams of using my AI/LLM skills to improve the AI functionality.  BUT… need such hacking skills as foundation.

Party on!

bm16ton, 2023-12-25:
Smart glasses like glasses you wear on your head with sum sorta hud display builtin? Ive heard rumblings but only ever seen in movies! Im a financially poor soul (no wprries rich in a thousand otherways) otherwise id be all over that! Feel free to contact me with any questions about general hacking on that front sounds amazing! Without the hardware ill be an admittedly limited resource but still id just be happy to know more/be involved in that!

rdaigle007, 2023-12-25:
Meta (formerly Facebook) has the most famous and much copied open source AI LLM.  Their smart glasses (made by Ray Ban) are $299.  Kind of ugly, but I like the tech.   It has cameras for taking photos or 1 minute videos.  And microphone and good quality speakers;  music sounds great and is neat to not need headphones,  but not loud enough so I have hard time hearing when walking on the street. Yes,  wear on the head glasses :)

think of the AI like a Google assistant or Apple Siri that you can ask questions to… but better than both of them.  And in Q1 next year you should be able to ask the AI about things you are looking at due to the cameras.  If they looked better on me,  I’d wear them all the time so a full-time AI assistant at my beckon call.

While I was an operating system developer 20 years ago,  my hacking skills are no where sufficient to even consider opening these glasses.  (I got prescription lenses,  so cost more than the $299 sunglasses version.)

Cool of you to offer assistance.  By Im 3 three thumbs with this type hacking.  

Oh,  no Augmented Reality display on this version.  In their next version they will do that…. It’ll be sick,  but of course much more expensive.  Not too mention the potential for sizzling the brain;  maybe that’s part of Meta’s plan to take over… AI take over the soft-tissue biological brain in these skulls.   *shrug*

bm16ton, 2023-12-25:
My god. Im gonna hafta google this and any other similiar devices. Cant
imagine they have much cpu horse power but things are always getting
smaller. Now im curious an mcu wich tethers to a cell phone etc? The only
reason i personaly wouldnt wear them or tge vr headsets is constantly
focusing on something about an inch away from my eyes my getting old eyes
is a recipe for dissaster.

On the wyze doorbell front i have usb mass storage and ext4 modules
uploaded, and along with backups. Im currently using a modded
system/init/forgot_the_name init which binds wz_mini_hacks to /opt plus a
handful of other things. I never provisioned mine so even tho it connects
to wifi it fails to authenticate with wyze so i disabled those services and
just started up the wz_mini iCamera binary then stupid holiday, had to
leave its untested. Telnets working, along with all other wz_mini standard
cli bins. Not sure how im gonna make it all installable without uart pins
being needed (like to try and make it ez for newcomers to diy hacking) im
currently binding a new /etc/passwd to bypass the unknown root password.
Looks like this doorbell with a much older firmware version uses sha512 the
shadow file in wz_mini (i believe is stock cam ver2) looks like maybe old
school blowfish? Looking like whatever i do in ill hafta work around that,
My only easyway out i see atm is if the init binary looks for any one of
the usb2uart devices and if found throws a login free root console on it.
And i was wrong earlier it looks like the zydas usb to wifi they added
support for isnt the old 802.11b/g ill hafta check but do believe its
actually the 802.11b only version!? Def a story here i just hope I
eventually get to hear it. And to make it slightly more difficult for
newcommers it wont turn on if powered by the AC inputs and it has its usb
otg id pin pulled(just having an otg inserted it locks hard immediatly)
Long ago a usb spec did exist for usb otg ports (non usb-c) to receive
power while in host mode. It was a some could do it some couldnt and the
ones that could would seldom had it implemented then i lost track i dunno
if it became the standard everyone used or not. Cannot remember what it was
called. Ill try and see if i can power it while its in host mode. But never
apply voltage to another set of pins in output mode, no protection at all.
Writing this has been the enjoyable part of this required yearly family
time lol.